z88dk/z88dk #1597: [testing targets] I am testing multiple targets. Which are known not to work at all as of today?

Fabrizio-Caruso, 2020-10-09:
I am testing multiple targets in order to have a complete picture of what works and what does not work.
I will share my results with you once I have a completed the tests. 

@zx70  @suborb 
Last time I checked a large number of targets I realized that:
- c7420
- m100
- x-07
do not produce a binary that can be run at all, i.e., nothing really happens when the binary is loaded and not even a hello world test can be run.

Is this still the case? Are these the only targets I can skip?

suborb, 2020-10-09:
No work has been done on those, so the state hasn't changed. I think +tvc is at an early stage as well.

Fabrizio-Caruso, 2020-10-10:
What about vgl? Does it have stdio.h with both printf  (or putchar) and getchar?

suborb, 2020-10-10:
+vgl is a newlib target, it should have printf and a getchar() equivalent.

It's not had any updates since contribution and I'm not sure where the docs is hiding (if any), so the machine may be due for inclusion into classic.

Fabrizio-Caruso, 2020-10-10:
Any other target not on the list in https://github.com/z88dk/z88dk/wiki/Platform I may consider for my tests?

zx70, 2020-10-11:
If you're hunting for new target, you could look for compatible models,  e.g. there could be a cp/m computer with special characteristics, like different terminal commands.   I suppose on cp/m the game needs to be turn based..

Fabrizio-Caruso, 2020-10-11:
@zx70 Generic CP/M targets with escape/control codes to control the cursor and getk working can be real-time.
If I compile my 3 Cross-Lib games for the generic  CP/M target, they will run on multiple CP/M machines (e.g., Kaypro2X) . I don't know how you guys managed to get getk to work in a generic way but it does work.

Fabrizio-Caruso, 2020-10-12:
I have now tested nearly all Z88DK (documented) targets.
I am missing CP/M boot images for:
- Coleco Adam
- BGR Computers Excalibur 64
- Epson QX/QC-10
- Sony SMC 77

Any hint?
Google has not helped me so far...

suborb, 2020-10-12:
I'll email you the discs this evening.

Fabrizio-Caruso, 2020-10-16:
TEST RESULTS
I have NOT tested all hundred targets and sub-targets with the very last snapshots but I have managed to get at least a working version of Cross Chase (and for many targets also Cross Shoot and Cross Bomber) on all targets (all documented targets + some undocumented such as vgl) except for 
- the 3 known broken (m100, c4720, x07)
- the untestable ones (no emulator such as rex and the "new retro" microboards), 
- the incomplete (s1mp3) and 
- pacman (which probably works but I have not yet figure out how to get it to work).

Very few targets lack proper real-time input an for those I just compile a turn-based version of the game.

The current list of targets (for all dev-kits) that can run Cross Chase are in: 
https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/docs/STATUS.md

closed 2020-10-16 by Fabrizio-Caruso