z88dk/z88dk #1598: [s1mp3] S1MP3 seems not to work at all

Fabrizio-Caruso, 2020-10-09:
If I try to compile a simple helloworld example I get:
```
/cygdrive/c/z88dk/bin/zcc.exe +s1mp3 \
./test/helloworld.c \
-create-app -o../build/hello_s1mp3.bin
appmake [+target] [options]

The z88dk application generator

This program is used to produce files which are suitable for use in
emulators or on the real hardware. Supported targets are:

+abc80        (abccas  ) - (C) 2000 Stefano Bodrato, (C) 2008 Robert Juhasz
+ace          (acetap  ) - (C) 2001 Stefano Bodrato
...
...
```

suborb, 2020-10-09:
There's no -create-app stage needed: https://github.com/z88dk/z88dk/wiki/Platform---S1MP3#compilation

For the sake of simplicity I should probably add a +Cz-noop to the config

closed 2020-10-09 by suborb

Fabrizio-Caruso, 2020-10-09:
Sorry! This is a stupid copy/paste mistake... I reuse Makefile targets.

Fabrizio-Caruso, 2020-10-09:
@suborb
Is any input function supported for this target? getchar does not seem to be supported, nor joystick.

If I try to compile:
```
#include<stdio.h>

int main()
{

    printf("Hello World!\n");
    getchar();
    return 0;
}
```
I get:
```
$ make hello_getchar_s1mp3
/cygdrive/c/z88dk/bin/zcc.exe +s1mp3 \
./test/helloworld_getchar.c \
-o../build/hello_getchar_s1mp3.bin
Error at file 'fgetc.c::fgetc' line 28: symbol 'fgetc_cons' not defined
Errors in source file c:/z88dk/\lib\target\s1mp3\classic\s1mp3_crt0.asm:
Error at file 'fgetc.c::fgetc' line 28: symbol 'fgetc_cons' not defined
make: *** [makefiles.chase/wip/Makefile.z88dk_wip_targets:243: hello_getchar_s1mp3] Error 1
```

suborb, 2020-10-09:
The simulator doesn't support any keypresses so I didn't add anything.

The real hardware does have buttons and this function (imported from s1mp3): https://github.com/z88dk/z88dk/blob/master/libsrc/target/s1mp3/keyboard/keyboard_getkeystate.c can apparently read them.

State of the world is here: https://z88dk.org/forum/viewtopic.php?f=3&t=11038

I think the summary is that this target is a source code preservation exercise (I rescued various bits from archive.org) and until someone comes up with some real hardware not much will move.