z88dk/z88dk #1601: [question/request][vblank] Wait for vertical blank without interrupts on targets that can do it?

Fabrizio-Caruso, 2020-10-11:
A common thing to do in games with many objects moving is to wait for vertical blank in order to reduce flickering or missing some animations. In order to make this work in plain C, I guess this has to be done assuming no interrupts are running.

What is the most portable way to do this on the highest number of targets?
Ideally a common API would be great but I guess this is a lot of work and it is impossible on some targets.

How can this be done on at least a large sub-set of the targets?
For example for targets with the VDP chip. Is there a way to do it for all such similar targets?

suborb, 2020-10-11:
The vdp machines have add_raster_int() - on sms there’s wait wait_vblank_noint()

On other machines halt is probably the best approach (via intrinsic.h)

On those without interrupts there will be no generic way of doing it.

Fabrizio-Caruso, 2020-10-11:
Thanks! I know that some systems have some sort of shadow registers in memory where you can read the scan line position. I thought some so simple could be possible in a generic way.

Fabrizio-Caruso, 2020-10-11:
Thanks a llot!

closed 2020-10-11 by Fabrizio-Caruso

Fabrizio-Caruso, 2020-10-23:
I don't understand why a raster interrupt is necessary on all VDP targets.
On the MSX one could just read location 0fc9e (aka JIFFY).
Maybe something similar can be done on all VDP systems. 
Don't VDP registers expose information on the raster position?

wait_for_vblank could be a common API across lots of Z88DK targets under some mild assumptions (default system interrupts running, default VDP settings, etc.).

suborb, 2020-10-26:
Given that you have a raster interrupt, it's trivial to achieve the same behaviour, something along the lines of:

```
#include <interrupt.h>

...
add_raster_int(tick_count_isr)
...

void wait_for_vblank()
{
#asm
    ld hl,_tick_count
    ld a,(hl)
loop:
    cp (hl)
    jr nz,loop
#endasm
}
```