z88dk/z88dk #1603: [feature][non -8x8 tiles] Could tiles be cut in a way to allow font data with the geometry of the target?

Fabrizio-Caruso, 2020-10-11:
I see that Z88DK uses 8x8 tiles only also non targets that have a different geometry. I understand the pros and cons of this choice.
So tiles are padded or cut to use 8x8 font data.
The way the cutting is done may make the target, more or less usable with specific font data.

There is a little problem for targets with narrower tiles because cutting the right-most column(s) makes it impossible to use font data with the hardware-specific number of columns.

If the cut is on the left-most (most-significant part), people could just use font data for that geometry without applying a shift.

Maybe there is a good reason why it was chosen to cut the right-most part.

suborb, 2020-10-11:
Can you remind which targets you're referring to?

Fabrizio-Caruso, 2020-10-11:
Super80r is such a target

Fabrizio-Caruso, 2020-10-11:
@suborb maybe there is a got reason for the current choice.
It would be nice to allow hardware-specific character width somehow. Cutting the most significant bits would make it possible.
For example in my case I already have 6x8 tiles for my universal games for ither non-Z88DK tagrgets (e.g., Oric 1, COMX-35, etc.) which I could re-use.

suborb, 2020-10-11:
Adding a SET_FONT_NATIVE is possible - we had a discussion at the time about the VG-5000.

The Super80r/v is a different case though - all z88dk does is write to the PCG area without any manipulation, so it's the MC6845 choosing which side to clip.

Fabrizio-Caruso, 2020-10-11:
Do you mean that I can already use native fonts?
I compile with --SET_FONT_NATIVE?

suborb, 2020-10-11:
No, it's not possible, but it's an option for me to add it in the future.

It wouldn't really help in this case since it's the display chip cutting off the right hand column.

Fabrizio-Caruso, 2020-10-11:
So, even if we could pass hardware-specific font data, in this very case, the lest significant bit would be lost anyway.

suborb, 2020-10-11:
Yes, in this particular case you'd have to pass a 9x7 character with the LSB of each byte unused.

Fabrizio-Caruso, 2020-10-11:
Thanks a lot for clarifying this.

closed 2020-10-11 by Fabrizio-Caruso