z88dk/z88dk #1604: [ondra][question][bug?] Ondra won't start

Fabrizio-Caruso, 2020-10-11:
I am trying my game(s) on the Ondra target.
I am trying to run on it on both ondrat and ondrav Mame targets. Is the Z88DK only meant for ondrav?
I am following the instructions on the ondra platform page.
I hit RETURN, attach the .wav and start the TAPE.
It loads and nothing happens.
Am I missing something?
What do I have to do to run the game?

suborb, 2020-10-11:
I only tested it on the ondrav. During loading, the numbers should increment and the program will auto-start.

The largest file I tested it on was about 14k - how big are your compiles coming out to?

Fabrizio-Caruso, 2020-10-11:
The tap and bin files are about 8k. I use the wav file with Mame.
I have attached the 3 files.

This looks wrong. Cross Chase should be bigger. Something may be wrong even before appmake.

[build.zip](https://github.com/z88dk/z88dk/files/5361430/build.zip)

Fabrizio-Caruso, 2020-10-11:
I am compiling with:
```
ondra:
	$(Z88DK_PATH)$(MYZ88DK) +ondra $(SCCZ80_OPTS) $(INCLUDE_OPTS) -vn -lndos \
	-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
	-D__ONDRA__ -DFORCE_XSIZE=40 -DFORXE_YSIZE=30  \
	-DCONIO -DNO_COLOR -DNO_INIT_GRAPHICS \
	$(FULL_FILES) \
	-create-app -o $(BUILD_PATH)/FULL_ondra.prg
	rm $(BUILD_PATH)/FULL_ondra.prg
```

Fabrizio-Caruso, 2020-10-11:
I just found a problem in my Makefile target...

Fabrizio-Caruso, 2020-10-11:
Sorry for opening this issue. It was a bug in the Makefile target.

closed 2020-10-11 by Fabrizio-Caruso