z88dk/z88dk #1606: [good news]ABC 800 can be easily tested thanks to Peter Anvin abcdisk tools

Fabrizio-Caruso, 2020-10-12:
ABC 800 has been very hard to test because I could not figure out how to create correct images since all of the tools I used had bugs.
It now works. I see only stdio.h is supported and it seems to work fine with Cross Chase in turn-based mode (only mode possible with plain stdio.h).

These tools seem now to be fixed thanks to the latest version of abcdisk by Peter Anvin (this guy: https://en.wikipedia.org/wiki/Hans_Peter_Anvin).

His tools are in:
https://www.zytor.com/pub/abc80/abcdisk/

I wonder whether they could be included in appmake. If not possible maybe we should mention them in the ABC 800 page to help people use Z88DK for this target as well as ABC 80.

suborb, 2020-10-12:
Could you write some words for the wiki page? 

I've not looked at the project/licence to see whether we could include it easily.

Fabrizio-Caruso, 2020-10-12:
Yes, tell me how to access the wiki page.
Peter may authorize you to include his code into appmake.
By the way his code is open. So it could be use anyway to understand the disk format.

suborb, 2020-10-12:
Thank you, you just be able to edit it - I've seen 3rd party contributions go in before.

Fabrizio-Caruso, 2020-10-12:
Peter may also help us test the generated code on real ABC80, ABC80X (ABC802 and ABC806).

Fabrizio-Caruso, 2020-10-12:
@suborb I have updated the Wiki page: https://github.com/z88dk/z88dk/wiki/Platform---Luxor-ABC800

Fabrizio-Caruso, 2020-10-14:
Peter told me that ABC800 understands esc/ctrl sequences natively but it is based on ADM3/A. So it may already be usable to do simple character graphics.

suborb, 2020-10-14:
Ooo, that's good, there is actually a driver for ADM3 in the CP/M library so it might be a "cheap and easy"  gencon solution once ported.

hpax, 2022-06-28:
To  be specific, the control sequences that the ABC80 and ABC800 console understand natively (this is assuming the console driver in ROM is called as opposed to writing directly to the display):

<ESC> = <x><y> 

... where <x> and <y> are [ASCII+32] positions the cursor

<FF>

... clears the screen

<BS>, <CR>, <LF>

... cursor movement; <LF> (including automatic <CR><LF> from printing past the end of the last line) scrolls the screen

<BEL>

... beep.

Adding +128 to any code keeps it from getting interpreted as a control code; instead it is stripped of the high bit and put in memory; this is used to create videotex graphics on ABC80, 800C, 802, and 806; 800M has no support for videotex.

The cursor is automatically lit when key input is expected through the console driver, and unlit when not.

The only special keys are <-, -> and Return which generate BS, TAB and CR, respectively; some ABC800 keyboards also have PF1-PF8 which generate PFn = 191+n+(shift*8)+(ctrl*16). The key combination Ctrl-< generates DEL, which is a printable character (large square bullet, per Videotex standard.)