z88dk/z88dk #1608: [mtx][bug] It seems that mtx (both mtx512 and mtx500) are no longer working

Fabrizio-Caruso, 2020-10-12:
I am compiling just:
 `
#include<stdio.h>

int main()
{

    printf("Hello World!\n");
    getchar();
    while(1){};
}
 `

with:
`
	$(Z88DK_PATH)$(MYZ88DK) +mtx -clib=ansi -startup=2 $(SCCZ80_OPTS)  \
	-lndos -create-app \
	./test/helloworld_getchar.c
 `

The output is a file "A" (no extension but in so called MTX format) .

If I load it I get a blank screen. Do I now have to switch to a special screen mode? 

I use Mess 0.118 (which is a version of Mame/Mess that could still load such files and can load old binaries produced with old versions of Z88DK).

I get this problem with and without -startup=2 and with and without POKE 64122,0 and  NEW commands.

Fabrizio-Caruso, 2020-10-13:
@suborb @zx70 
I confirm that we have something that looks like a  regression on the MTX. It does work with snapshot 2018 09 21.
I get a working binary (except for 2 vissible garbadge hardware sprites) if I compile my code with Z88DK 2018 09 21.

Fabrizio-Caruso, 2020-10-13:
P.S.: The newest versions of Mame can also load the .wav files... I am retesting it. I may update the docs if I can confirm it.

suborb, 2020-10-13:
-startup=2 corresponds to a ROM image, omitting that creates files that load and run correctly.

Fabrizio-Caruso, 2020-10-13:
The doc says
-subtype=rom is the option for ROM
-startup=2 is the option for MTX512 (no such option creates binaries for MTX500)
and that is how it seems to work on older snapshots.

How do I create MTX500 vs MTX512 images?

suborb, 2020-10-13:
I think this image will help both of us!

[image]

The default config creates the binary from 32k onwards. This works well for both MTX512 and MTX500.

The old -startup=2 (not ROM) generates code from address 16k onwards, however this doesn't seem to work - it definitely won't work on the MTX500 since there's no memory there.

Fabrizio-Caruso, 2020-10-13:
@suborb ok, I understand that the old -startup=2 now has a different meaning.
Anyone I can no longer build a working .MTX binary even if I do not use -startup=2. If I use an old snapshot it does work.

suborb, 2020-10-13:
Thanks for the binaries!

If you look at them with a hex editor, you'll see (towards the start of the file probably 1k-ish in) (32787) in the 2018 one, and (16403) in the one from today.

If I compile with no startup option I see (32787), if I compile with -startup=2 I see (16403).

So, I think you've got a -startup=2 option creeping in somewhere. This may well be due to a lingering zcc_opt.def file? Or something in your makefiles?

Fabrizio-Caruso, 2020-10-13:
Are you saying that the new meaning of startup is what is breaking my builds?

So HTron must also be broken as RobertK in the latest version of April 2020:
REM MTX 500 mode (32K)
REM This memory model works with the MTX512 too, but before loading the user must type:
REM POKE 64122,0 
REM NEW
zcc +mtx -pragma-redirect:fputc_cons=fputc_cons_generic -create-app htron.c -o htron_MemotechMTX500.o
ren htron_MemotechMTX500 htron_MemotechMTX500.mtx

REM MTX 512 mode (64K and more)
zcc +mtx -pragma-redirect:fputc_cons=fputc_cons_generic -create-app -startup=2 htron.c -o htron_MemotechMTX512.o
ren htron_MemotechMTX512 htron_MemotechMTX512.mtx

Fabrizio-Caruso, 2020-10-13:
@suborb if I use startup=2 with the old snapshot, I get a working game that does not require POKE 64122,0 on the MTX512.

Fabrizio-Caruso, 2020-10-13:
I delete zcc_opt.def file
I compile with no startup=2 and I do not see any start-up if even if I enable -v.

Maybe I get an implicit  startup=2 through some other options:
```
	$(Z88DK_PATH)$(MYZ88DK) +mtx $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
	-D__MTX__ -DFORCE_SCREEN_MODE=2 \
	-create-app -oMTX.bin -v \
	-DFULL_GAME -DSOUNDS  -DEND_SCREEN -DBETWEEN_LEVEL \
	-DCONIO \
	-lndos -DSOUNDS -DFORCE_PSG \
	-DANIMATE_PLAYER \
	-DUDG_GRAPHICS \
	-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES) \
	$(CROSS_LIB_PATH)/sound/z88dk/psg/psg_sounds.c
```

Fabrizio-Caruso, 2020-10-13:
@suborb could you just tell how to compile a simple hello world with today's snapshot?

suborb, 2020-10-13:
I'm total baffled by this one.

I thought I was able to reproduce the problem. But I've lost the knack: I've gone back to the build from last night and I still can't reproduce it.

Working through examples/console

zcc +mtx world.c -startup=2 -create-app
zcc +mtx -clib=ansi ansitest.c -startup=2 -create-app

And for me, both of them work.

Fabrizio-Caruso, 2020-10-13:
How are you testing your examples? Mame? Memu? Which versions?

suborb, 2020-10-13:
I'm using MEMU, the binaries have a date of 2017-10-21

Fabrizio-Caruso, 2020-10-13:
How do you load them?
Just LOAD"A"
No need for POKE 64122,0 NEW?

suborb, 2020-10-13:
Just LOAD "" then RUN

Fabrizio-Caruso, 2020-10-13:
Maybe if you remove -startup=2 they won't work unless you type POKE 64122,0 NEW before LOAD""

Fabrizio-Caruso, 2020-10-13:
@suborb, I can reproduce helloworld with startup=2 and it works.
Without startup=2 does *not* work as expected (as in the old doc).
But no startup=2 requires POKE 64122,0 NEW before LOAD""

Fabrizio-Caruso, 2020-10-14:
So we need whatever startup=2 was doing.
I also have the problem with my game as it no longer starts with or without startup=2 with recent Z88DK versions.

suborb, 2020-10-14:
Yes I can see there’s some issue with some of the examples. I’m trying to figure out what it is - the obvious problem is size though I’ve not found the “cutoff” point yet. 

The odd thing is that the mtx code hasn’t changed in a long time. So it’s a bit puzzling.

Fabrizio-Caruso, 2020-10-14:
@suborb Moreover startup=2 (snapshot 2020 10 13 and before) behaves as previously documented (for binaries that are still working such as helloworld).
I am testing with SNAPSHOT 2020 10 13 because you may have changed the behavior of startup=2 last night.

I have the following expected results that are coherent with the previous doc:
HELLO500 without startup=2 on MTX 500 LOAD"" -> OK
HELLO500 without startup=2 on MTX 512 LOAD"" -> CRASH
HELLO500 without startup=2 on MTX512 POKE64122,0:NEW LOAD"" -> OK

HELLO512 with startup=2 on MTX500 -> CRASH
HELLO512 with startup=3 on MTX512 -> OK

So startup=2 was useful for bigger games which only run on MTX512 and you want to avoid to type POKE 64122,0:NEW

Summary:
startup=2 -> MTX512 only and no need to type POKE 64122,0:NEW
no startup=2 -> MTX500 just LOAD"" or MTX512 but you need to type POKE 64122,0:NEW before LOAD""

I have attached the two binaries for you to reproduce.
In order to reproduce it I use Bill's version of MEMU (http://www.primrosebank.net/computers/mtx/emulators/Bills_MEMU_ds.htm) and I **always** close and restart the emulator because a simple restart is not enough. Bill's version has a menu (F11) to select the .MTX file but restart won't clean everything.

[HELLO.zip](https://github.com/z88dk/z88dk/files/5377331/HELLO.zip)

suborb, 2020-10-14:
I switched to testing in mame to get hold of a debugger....and I've made a few discoveries.

In general, to do not call the ROM printer with any non ascii characters - you end up being turfed back into basic - this was the breakage I was seeing a lot of. Completely unrelated to yours though.

And regarding cross chase. I think there's a screen mode switch missing - the MTX is running in the 40 column mode and you want graphics in mode 2, bodging that in to a local compile fixes the problem. I'm guessing the old version of z88dk (which didn't support multiple screen modes on the VDP) just switched to mode 2 on startup.

Fabrizio-Caruso, 2020-10-14:
I thought I was forcing screen mode 2... Gonna recheck.
Can I use today's snapshot with startup=2?

suborb, 2020-10-14:
Yup, I left it in. The only real change is that the mtx is now created with a mtx extension by default.

Fabrizio-Caruso, 2020-10-14:
Indeed I forgot to force the screen mode. X Chase starts but crashes soon after (tested on MEMU):

[image]

suborb, 2020-10-14:
That seems to happen when the player dies? What do you do then? Play a sound? Something else?

Fabrizio-Caruso, 2020-10-14:
It seems to happen very quickly... Let me test it again.

Fabrizio-Caruso, 2020-10-14:
Die is one cause for the crash. Testing for other causes...

Fabrizio-Caruso, 2020-10-14:
Ok, it is the sounds.

Fabrizio-Caruso, 2020-10-14:
I am using PSG sounds.

suborb, 2020-10-14:
I'm not sure how I've missed this, but the snXXXXX code wasn't cleaned for for each target. So in this case, the MTX ended up writing to port 0xff. In MEMU this triggers an NMI call which presumably launches a monitor or something.

I can't actually get any sound to be heard, but at least it's not crashing anymore.

Fabrizio-Caruso, 2020-10-14:
Bit-bang does not work but does not crash anything.
I have also realized that getk does not have auto-repeat...

closed 2021-01-03 by suborb