z88dk/z88dk #1609: [vgl][info/request] VGL works! Easily tested with gl4000 mame target

Fabrizio-Caruso, 2020-10-13:
The VGL target works fine.

TEST:
I have only tested stdio.h input/output.
Testing this target is very simple and requires no rom hacking.
Just use mame gl400 target and select the cartridge. These toy computers maybe had some sort of cartridge slot.

SUGGESTIONS:
It should probably be inserted in the list of available targets.
I hope it will get classic lib support one day.
It would be nice to have more than 4 rows by using smaller fonts. Even 2 extra rows would make it more usable.

Is there any getk equivalent function ?

suborb, 2020-12-22:
I'm tinkering around with this. There's nothing that can be done about the display size - the screen is a hd44780 so it's fixed at 4x20 in the gl4000 model.

The 6000 series are probably a bit more interesting in terms of usable target.

Fabrizio-Caruso, 2020-12-23:
@suborb Thanks! I will test it. I am able to run a sort of playable turn-based version of XChase on the 4x20 screen but 6 or 8 rows would help a lot with more games. Do these toy computers have a bitmap mode or just text mode?

suborb, 2020-12-23:
It's just text mode - the hd44780 does support 8 (I think) PCG characters but I've not investigated yet.

To be honest, with the 2000, 4000 I'm probably not going to add too much more support: they're just a bit too limited to be useful IMO.

I'm looking into the 6000sl which supports a 240x100 pixel screen which is a lot more interesting as a general purpose target.

Wiki page will be forthcoming when I'm happy with where I've got to, but if you want to play with the cutting edge, feel free!

Fabrizio-Caruso, 2021-04-01:
@suborb, Thanks I will use it in Cross-Lib and add it to the list of targets I can support (which for the Z80 targets it is thanks to Z88DK)

closed 2021-05-01 by suborb