z88dk/z88dk #1612: [coleco_adam][psg] Hissing sound never stops

Fabrizio-Caruso, 2020-10-14:
When I use PSG sounds I get a hissing sound that never stops.

suborb, 2020-10-15:
I need to set up a issue template, however could you:

a) Provide a short, self-standing example of the problem
b) Provide the command line you use to compile

Fabrizio-Caruso, 2020-10-15:
@suborb Sorry. You are perfectly right. This issue is not specific at all. I will provide more details.

Fabrizio-Caruso, 2020-10-15:
You can reproduce the hissing sound with the following code:
```
#include <psg.h>
#define A_PERIOD_LOW 0
#define A_PERIOD_HI 1
#define B_PERIOD_LOW 2
#define B_PERIOD_HI 3
#define C_PERIOD_LOW 4
#define C_PERIOD_HI 5
#define NOISE 6
#define CONTROL 7
#define A_VOLUME 8
#define B_VOLUME 9
#define C_VOLUME 10
#define ENV_PERIOD_LOW 11
#define ENV_PERIOD_HI 12
#define ENV_WAVE 13
void ZAP_SOUND(void) {
	uint8_t i;
	uint8_t j;
	set_psg(A_VOLUME,15);
	set_psg(A_PERIOD_LOW,255);
	set_psg(A_PERIOD_HI, 15);
	set_psg(CONTROL, 0xFF  - 0x01);
	for(i=0;i<16;i++) {
		set_psg(A_PERIOD_HI,15-i);		
		for(j=0;j<150;++j){}
	}
	set_psg(A_VOLUME,0);		
}
int main() {
    ZAP_SOUND();
    while(1){};   
    return 0;
}
```
which I compile with:
```
coleco_sound_test: 
	$(Z88DK_PATH)$(MYZ88DK) +cpm -subtype=adam -lndos -create-app  -vn ./test/psg_test.c
```

Sorry for not providing this example before.

suborb, 2020-10-15:
Thank you.

Now, I'm *really* confused as to what you're doing! 

    extern void __LIB__ set_psg(unsigned int reg, unsigned int val) __smallc;

So if we take this line:

    	set_psg(A_VOLUME,15);

The intention is to set register A_VOLUME (8) to value 15.

However, looking at the data sheet:

http://map.grauw.nl/resources/sound/texas_instruments_sn76489an.pdf

The data writes take this form:

[image]

With the R fields:

[image]

Which really doesn't match what you're intending to do, but does make sense of what you're observing/hearing!

Fabrizio-Caruso, 2020-10-15:
:-(
Ok, so I have done a big mess...
I have to restudy those tables to understand what I have to do. It sort of works though. So I got something right and something totally wrong.

closed 2020-10-15 by Fabrizio-Caruso

Fabrizio-Caruso, 2020-10-15:
I am lost because I don't understand what is wrong with the way I set the value.

From https://www.cpcwiki.eu/index.php/PSG#07h_-_Mixer_Control_Register
it seems that setting register 8 to 15 sets the volume to max and 0 to min.
[image]

suborb, 2020-10-15:
Those docs are for the AY819x chip. The Adam (&Pencil2 etc) have an SN76489

Fabrizio-Caruso, 2020-10-15:
@suborb I will stick to bit-bang unless I understand PSG...
I thought all PSG were programmed similarly. So if I want to use psg_set on the CPC do I have to follow a different semantics?

suborb, 2020-10-15:
The CPC has got an AY819x, so the code you've got above should work correctly on it!

Fabrizio-Caruso, 2020-10-15:
@suborb so maybe the interface for AY819x should not be the same as the one for SN76489 if we want to have portable code.
In my ignorance I thought the PSG covered all PSG-cases with some sort of abstraction. I was very wrong. :-(

Fabrizio-Caruso, 2020-10-15:
Indeed it works on CPC... So psg is not meant as an abstraction layer but it is writing into different registers that on each chip have a different meaning. I am sorry I missed this important point.