z88dk/z88dk #1617: [laser500][appmake] WAV on real hardware do not work (but can be fixed)

Fabrizio-Caruso, 2020-10-22:
I am reporting an issue that @LoudScotsBloke can reproduce on real hardware.

The wav files as produced by appmake won't load on real hardware but @LoudScotsBloke has reported that he is able to "fix" them by the following manual procedure:
1. Change to 22KHZ
2. Apply a lo-pass filter @ 8KHZ
3. Amplify by 20Dbs

nippur72, 2020-10-24:
last time I checked it worked on my Laser 500. It was long time ago but I don't think anything has changed since then.

I guess it's an issue related to volume. To avoid such problems I use a "car-cassette adapter" in my tape recorder and feed the audio directly from the PC earphone plug or other source. It works nicely.

@LoudScotsBloke, if you want to experiment with Laser 500 and .wavs I have written an utility that compresses Laser binaries with "exomizer" and then adds a custom turbo-tape with various speed settings that can load programs in seconds instead of minutes. Higher speeds won't work when recorded on a real tape though, but they are ok when played from PC.

The utility is [here](https://github.com/nippur72/laser500-wav) and is also available as `npm` package `laser500-wav` - write me if you have troubles running it.

BTW, changing to 22 KHz just degrades the audio signal, and the +20 dB amplify exceeds the maximum volume. `appmake` generates audio at 75% level so boosting it by +20 dB results in audio at 100%. Anyway no distortion is introduced because the source is a pure square wave that doesn't distort no matter how much you amplifiy it.

zx70, 2020-10-24:
sometimes I put in an option,  -Cz--loud to get a higher audio level

zx70, 2020-11-03:
..by the way I added a new option, -Cz--22   😏

closed 2020-11-03 by zx70