z88dk/z88dk #1619: add support for LM80C computer

nippur72, 2020-10-24:
LM80C is a modern Z80 home computer built with vintage chips, all designed by Leonardo Miliani @leomil72. It would be nice if it could be added to Z88DK. 

Below I summarized the basic information that are needed for a minimum support. The machine itself can be seen as a simpler version of the MSX1 or Spectravideo. 

## LM80C Hardware

- Z80 3.5 MHz
- Video TMS9918 (NTSC) with 16K VRAM
- Audio AY38910
- 32K RAM
- 32K ROM with custom Microsoft BASIC, latest firmware is R3.14
- Z80 SIO, PIO and CTC
- Commodore C16 repurposed keyboard
- No joysticks
- No storage device (see "misc" below)

## Useful URLs

- online emulator: https://nippur72.github.io/lm80c-emu
- author's repo with sources and schematics: https://github.com/leomil72/LM80C
- FPGA implementation for the MiST: https://github.com/nippur72/LM80C_MiST

## 64K Z80 ADDRESS SPACE MAP

```
$0000-$7FFF Firmware ROM
$8000-$8240 BASIC work space
$8241-$FFFF Free RAM (32074 bytes available to BASIC)
```

## I/O ports

```
$00-$03   Z80 PIO (parallel interface)
$10-$13   Z80 CTC (counter and timer)
$20-$23   Z80 SIO (serial interface)
$30       TMS9918 data port
$32       TMS9918 register port
$40-$43   AY-3-8910 (audio)
```

## MISC

1) On boot, CTC generates an INT interrupt at 100Hz rate for updating
the clock and reading the keyboard.

2) Keyboard is connected to the parallel ports of the audio chip,
CPU queries keyboard columns on port A and receives the rows on port B
(see `KEYBOARD` rom routine).

3) The system currently has no storage device (save/load from SD card is
in development). On the real machine programs are loaded via serial
interface (BASIC programs). On the emulator, you can drag & drop a ".prg"
file on the emulated screen and then RUN it.

4) the builtin font is standard ASCII extended to 255 with petscii-like
graphics. There is no reverse.

## USEFUL ROM ENTRIES

```
0C88: KEYBOARD
06C8: CHAR2VID
```

a complete ROM .lst file can be found here:
https://github.com/nippur72/lm80c-emu/blob/master/rom.lst

## BASIC LAUNCHER

C programs can be launched by adding the following header
to the ".prg" binary file output:

```
0x51, 0x82, 0xe4, 0x07, 0xab, 0x26, 0x48, 0x38,
0x32, 0x35, 0x33, 0x3a, 0x80, 0x20, 0x20, 0x00,
0x00, 0x00
```
which results in the following BASIC stub program:

```basic
2020 SYS&H8253:END
```
normally loaded at 0x8241:
```
8241: 51 82 e4 07 ab 26 48 38
8249: 32 35 33 3a 80 20 20 00
8251: 00 00

8253: /* c program follows here */
```

Compiled program can be tested on the emulator by simply drag&dropping
the ".prg" file on the emulator window (and then RUN + enter).

suborb, 2020-10-26:
Apart from the keyboard I think that this should be an easy one to add - I'll look at the bios sources and see if there's some convenient keyboard reading routines to make it even easier.

suborb, 2020-10-26:
I think I've added enough for the Cross games to work now. I do want to add support for the <input.h> which will allow multiplayer games to be supported - just remember to switch to mode 2 before doing anything.

@nippur72 thanks for the information, really helpful. BTW, is rom.rom derived from rom.lst? Some of the entry points for the MBF32 library look they've moved between the listing and the used ROM.

nippur72, 2020-10-27:
@suborb sorry it's an old file, the official ROM file is `LM80C-firmware-r314.rom` from which I derived the `.lst`. Sources are also available on the computer's repo in separate `.asm` files.

The firmware is quite stable now, sometimes when a new release comes out the pointer to the start of BASIC gets changed and some things need to be updated.

leomil72, 2020-10-31:
Hello, I'm the original developer of the system. I apologize I didn't introduce me before, but I'm really busy.... I was informed about this support and read about the issue involving the BASIC pointers... Since the last firmware release (R3.14) is almost complete and stable, for your benefit I could freeze the development of the system in a few relaeses... I only need to add support for I/O routines to read/write files from an external SD card via a serial port. But at the moment I can not predict what the impact of the code of such statements will be on the firmware, and eventually where the BASIC pointers will be moved to.

suborb, 2020-10-31:
Hi @leomil72 not a problem. During normal operation the only bit of ROM related code we use is the value written to LASTKEYPRSD = $822c, though I hope to add non-ROM based input at some point. All the display code is handled by custom libraries for portability reasons.

The RS232 library uses rst $08 and rst $10 which I guess will never move?

The issue with moving pointers came about when I added support for the maths package in the ROM, again it's not really an issue since we can just use one in application space. However, if you could find a stable space for a jump table that would be helpful (typically a full fp package is a couple of k, so using a ROM based can help when memory is tight) - the functions we call are here: https://github.com/z88dk/z88dk/blob/master/lib/target/lm80c/def/maths_mbf.def

suborb, 2021-05-01:
Adding inkey keyboard support would be nice, but this target is functional as it stands.

closed 2021-05-01 by suborb