z88dk/z88dk #1621: Add stdcbench results at https://github.com/z88dk/z88dk/wiki/Benchmarks?

spth, 2020-10-28:
At https://github.com/z88dk/z88dk/wiki/Benchmarks, a few benchmarks are used to compare compilers.

Due to the shortcomings of Dhrystone, Whetstone, Coremark and the unsuitability of major benchmarks, such as SPEC for small systems, I wrote stdcbench:

https://dl.acm.org/doi/abs/10.1145/3207719.3207726
https://sourceforge.net/projects/stdcbench/

stdcbench is one of the benchmarks used to track code size and speed over SDCC revisions:

https://sourceforge.net/p/sdcc/code/HEAD/tree/trunk/sdcc-extra/historygraphs/stdcbench-stm8-size.svg
https://sourceforge.net/p/sdcc/code/HEAD/tree/trunk/sdcc-extra/historygraphs/stdcbench-stm8-score.svg

How about adding stdcbench results at https://github.com/z88dk/z88dk/wiki/Benchmarks?

suborb, 2020-10-29:
I've started to take a look, it's showing up a few edge cases that I need to handle better in sccz80.

So far the conclusion is (to no great surprise), that loops in sccz80 aren't that great.

suborb, 2020-10-30:
Because I'm running under ticks the number produced is a measure of fast (well slow!) my laptop is, so I've made a minor tweak to just run through the benchmarks once - I've only got c90base to work with zsdcc (4.0.3 #11877) at present unfortunately. Sadly this means we can't easily compare vanilla sdcc and z88dk:

`zcc +test -compiler=sccz80  -O2  -m`  -> 173_691_034 ticks, 9417 bytes
`zcc +test -compiler=sccz80  -O2  -m --opt-code-speed=all`  -> 145_485_000 ticks, 9599 bytes
`zcc +test -compiler=sccz80  -O2  -m --opt-code-speed=all -lfastmath` -> 122_275_012 ticks, 11877 bytes
`zcc +test -compiler=sccz80  -O2  -m -clib=8080`  -> 168_723_490 ticks, 9612 bytes
`zcc +test -compiler=sccz80  -O2  -m -clib=rabbit`  -> 56_163_266 ticks, 8632 bytes
`zcc +test -compiler=sccz80  -O2  -m -clib=gbz80`  -> 170_860_224 ticks, 9461 bytes

`zcc +test -compiler=sdcc -O2  --max-allocs-per-node100000 --opt-code-speed -m` -> 94_211_835 ticks, 8749 bytes
`zcc +test -compiler=sdcc -O2  --max-allocs-per-node100000 --opt-code-speed -m -lfastmath` -> 86_213_355 ticks, 11041 bytes
`zcc +test -compiler=sdcc -O2  --max-allocs-per-node100000 --opt-code-speed -m -clib=rabbit` -> 49_282_502 ticks, 8320 bytes

There's obviously something wrong with the first sccz80 test - I'm not sure how the 8080 has managed to outpace it.

Given the lack of a context-aware optimiser (or indeed any code analysis) in sccz80 and the difference in compilation time (on my machine sdcc is taking 47 seconds, sccz80 takes just 3 seconds) it's not as bad as it could have been.

Any long tests will be interesting: the sccz80 generated code for longs is much more compact - see https://github.com/z88dk/z88dk/wiki/WritingOptimalCode#sample-results

feilipu, 2022-05-11:
@suborb Did you publish the modifications required to get STDC Bench to compile with z80? I had a fleeting go but, didn't get far.

It would be nice to add at least some compiler runs for this benchmark.

feilipu, 2026-10-27:
stdcbench 0.8, Z80, 4 MHz, 27 Oct 2026.
Bytes: both-module iotimer `.bin`, page zero included.
Score: TIMER formula, one decimal. Printed: that value truncated.
80cc: no `-fframe-pointer`. zsdcc: 4.6.0 #16639.
These rows are 0.8. The paper in this directory measured 0.4.

**Classic** (`+test`)

| Compiler | Bytes | c90base | c90lib | Total | c90base ticks | c90lib ticks | Printed |
|----------|------:|-------:|-------:|------:|--------------:|-------------:|---------|
| sccz80 | 27605 | 1.9 | 3.3 | 5.2 | 168,749,272 | 491,595,488 | 1, 3, 4 |
| 80cc | 28110 | 3.1 | 3.9 | 7.0 | 102,158,431 | 410,457,676 | 3, 3, 6 |
| zsdcc | 25341 | 3.5 | 4.4 | 7.9 | 91,051,880 | 367,241,838 | 3, 4, 7 |

**Newlib** (`+z80 -startup=0`)

c90lib rows use the fast spare-tail allocator (`__CLIB_OPT_MALLOC = 1`).
c90base does not call malloc. Its tick count excludes heap init.

| Compiler | Bytes | c90base | c90lib | Total | c90base ticks | c90lib ticks | Printed |
|----------|------:|-------:|-------:|------:|--------------:|-------------:|---------|
| sccz80 | 29010 | 1.9 | 2.9 | 4.8 | 168,749,224 | 557,117,716 | 1, 2, 3 |
| 80cc | 29178 | 3.1 | 3.4 | 6.5 | 102,158,431 | 475,976,814 | 3, 3, 6 |
| zsdcc | 26691 | 3.5 | 3.8 | 7.3 | 92,368,035 | 419,437,901 | 1, 1, 2 |

Newlib zsdcc printed 1, 1, 2 from the `__div32` call. The row is the TIMER formula. No ERROR line.

closed 2026-10-27 by feilipu