z88dk/z88dk #1626: (build) Migrating away from travis

suborb, 2020-11-03:
Nothing is for free, and waiting for travis to build isn't costless either.

Since the recent changes, the Linux build queue for travis is regularly 5-6k jobs which results in a very long wait time.

Need to take a look at the alternatives, circleci, travis, cheeky gitlab mirror, GitHub actions. I'm not adverse to running an agent locally on my vm server as long as it can be secured.

suborb, 2020-11-04:
I've had a horrible time with the perl requirements for the z80asm tests - I just can't get the right magical combination that doesn't fail with an annoying error, as a result I've ignored them @pauloscustodio could you take a look?

I've setup both Github actions and Azure pipelines. Actions is limited to 2000 minutes/month and with each build taking (max 15 minutes) that's 133 builds/month, 4 per day.

I think Azure doesn't have a limit, so I'll leave that running as a backup/secondary CI system. Build status is here: https://dev.azure.com/z88dkorg/z88dk (someone else seems to have the z88dk name which is a smidgen annoying).

I'll decommission the travis build since it's not much use as a CI system due to the queue times being so long.

pauloscustodio, 2020-11-06:
I'll have a look.

pauloscustodio, 2020-11-06:
I have fixed the problem in Github Actions (still not merged).
@suborb, where do you setup the Azure build?

suborb, 2020-11-06:
It's in azure-pipelines.yml - similar to actions, but I'm not sure that environment variables work in the same way.

PR #1638 (2020-11-10): Z80asm issue 1626 3
Fix #1626 - build in Github actions and Azure Pipelines
closed 2020-11-10 by pauloscustodio