z88dk/z88dk #1631: Question about creating libraries.

jayacotton, 2020-11-04:
I have a large number of subroutines written in C and each as its own file.

I'd like to use ar and ranlib to handle the binaries, but that seems to not be working.

What is the 'correct' way to build a linkable library from this pile of binary ?

suborb, 2020-11-05:
The 'correct' way is:

z80asm -xlibraryname files

Where files is either a list of files, or @listfile.lst which contains a list of files to include.

jayacotton, 2020-11-14:
Sorry, I guess I did not give you enough context.  

Using zcc 
CFLAGS = -O --list --c-code-in-asm -Wall -xplot10.lib

# compile, then strip unnecessary symbols
.c.o:
        $(CC) +rc2014 -DSkip_f2c_Undefs $(CFLAGS) -c $*.c

here is the segment from the zcc --help
   -S --assemble-only           Stop after compiling .c .s files to .asm files
   -x                           Make a library out of source files
      -create-app               Run appmake on the resulting binary to create emulator usable file

This produces errors.

suborb, 2020-12-21:
Sorry, your update hit whilst I was having a little break.

I've just tried out the following:

````
zcc +test -x -o mylib *.c
````

And a library file (mylib.lib) was produced. I think the key difference between my line and yours is that the -x doesn't take any option and uses the value of -o as the basename of the library file.

jayacotton, 2020-12-21:
that does what I needed.

tnx
jc

closed 2020-12-22 by suborb