z88dk/z88dk #1633: Please add a home-brew package

diogoeichert, 2020-11-08:
As a macOS user, it would be convenient to be able to install z88dk with:
```
brew install z88dk
```

suborb, 2020-11-08:
Reading: https://docs.brew.sh/Acceptable-Formulae

A couple of problems:

- nightly builds are not welcome 
- “We frown on authors submitting their own work unless it is very popular”

diogoeichert, 2020-11-08:
That looks tough. Maybe using a [tap](https://docs.brew.sh/Taps#the-brew-tap-command) instead of core? 🤔

desertkun, 2023-01-21:
700 stars though. And we could be fine w/o pushing a nightlies, having regular releases there would be extremely good to.

desertkun, 2023-01-22:
Anyhow I took a look at the [guide](https://docs.brew.sh/Acceptable-Formulae) and it seems worth a try.