z88dk/z88dk #1634: [question/bug?][spectrum] Spectrum and stdio.h

Fabrizio-Caruso, 2020-11-08:
Is `stdio.h` supposed to work with default compiler options on the spectrum target (and most other targets)?

If I compile my code in:
https://github.com/Fabrizio-Caruso/Tiny-RPG
which only uses `stdio.h` for input/output I get no scroll up and glitches (see image) when `printf` reaches the last line.

I compile with:
```
spectrum_48k:
	$(Z88DK_PATH)$(MYZ88DK) +zx $(INCLUDE_OPTS) \
	-lndos -create-app \
	-o ...
```

[image]

suborb, 2020-11-08:
Can you reproduce this with adv_a.c? I can't at the moment.

That program just uses stdio so should be a comparable testcase - press return a few times to get the screen to scroll.

Fabrizio-Caruso, 2020-11-08:
This is enough to reproduce the problem:
```
#include <stdio.h>

int main(void)
{
    while(1)
    {
        printf("Hello %s\n", "world");
        getchar();
    }
	return 0;
}
```

Is `getchar()` supposed to work by default?

suborb, 2020-11-08:
I still can't reproduce it: I first thought it was a ts2068/zx issue but that's autodetected these days. I'm using fuse as the emulator BTW.

Can you provide the compile line and the z88dk version you're using? Not much has changed lately, but you might have a build where I broke everything by accident!

Fabrizio-Caruso, 2020-11-08:
I am using the very latest Z88DK snapshot and testing with Fuse 1.3.7

I have attached the .tap produced by Z88DK with appmake.
[hello_spectrum.zip](https://github.com/z88dk/z88dk/files/5507002/hello_spectrum.zip)

[image]

suborb, 2020-11-08:
Ah, pressing enter, I was pressing other keys!

I'm guessing we're just running off the bottom of the screen and the bounds checking is failing, I'll take a look.

closed 2020-11-08 by suborb

Fabrizio-Caruso, 2020-11-09:
I confirm that the issue is fixed on the spectrum target but a similar issue (extra newline characters) is not fixed on the cpc target.

Fabrizio-Caruso, 2020-11-09:
@suborb The issue cannot be reproduced by reducing my code to a smaller size.
@suborb Something fishy is going on on the CPC target.
If I keep the *very* same main() and remove all *unused* other functions, the error no longer appears.
If I re-add the functions, I see the error re-appear.
I was thinking of the stack but: ` -pragma-define:REGISTER_SP=-1` does not fix the problem.

KO: (https://github.com/Fabrizio-Caruso/Tiny-RPG/blob/main/src/games/conan/main.c)
[image]

OK: (https://github.com/Fabrizio-Caruso/Tiny-RPG/blob/main/src/test/test.c)
[image]

commit 06130610 (2020-11-08): Fix #1634: Bounds checking, get it right, hopefully...