z88dk/z88dk #1637: [cpc][stdio.h] scroll up in printf prints each next character in a newline

Fabrizio-Caruso, 2020-11-08:
This issue may be similar to a recent issue on the spectrum target that @suborb has just fixed #1634 

Example:
```
#include <stdio.h>

int main(void)
{
    while(1)
    {
        printf("Hello %s\n", "world 8765786578657854875487548756486548654765486548765478654865");
        getchar();
    }
	return 0;
}

If I press return multiple times untill a scroll up is produced, then the printf starts to display characters as shown on the picture below.

cdt and dsk files:
[build.zip](https://github.com/z88dk/z88dk/files/5507286/build.zip)

![image](https://user-images.githubusercontent.com/29057615/98485967-9210ca00-221a-11eb-8e8a-f1224ddfd82c.png)

suborb, 2020-11-08:
I think it is the same issue, the code I fixed was generic so affects every target.

I've no idea how I missed this since I always run adv_a.c to test scrolling :(

Fabrizio-Caruso, 2020-11-08:
@suborb I cannot reproduce the problem on the MSX.

Fabrizio-Caruso, 2020-11-09:
I am closing this because it only appears on Caprice but not on JavaCPC emulator. I will retest it on both emulators.

Fabrizio-Caruso, 2020-11-09:
@suborb I confirm this issue that I can now reproduce on both Caprice and JavaCPC emulator.

I have used the very latest snapshot, which fixes the problem I reported on the Spectrum but does not fix it on the CPC:

[image]

Fabrizio-Caruso, 2020-11-09:
@suborb I am trying to construct a simpler example than my text game code.
For the moment I can say that the problem seems to come from:
```
    printf("%s - life: %u - stamina: %u - strength: %u - dexterity: %u\n", 
           character_ptr->name, 
           get_life(character_ptr), get_stamina(character_ptr),
           get_strength(character_ptr), get_dexterity(character_ptr));
```

Fabrizio-Caruso, 2020-11-09:
@suborb The issue cannot be reproduced by reducing my code to a smaller size.
@suborb Something fishy is going on on the CPC target.
If I keep the *very* same main() and remove all *unused* other functions, the error no longer appears.
If I re-add the functions, I see the error re-appear.
I was thinking of the stack but: ` -pragma-define:REGISTER_SP=-1` does not fix the problem.

KO: (https://github.com/Fabrizio-Caruso/Tiny-RPG/blob/main/src/games/conan/main.c)
[image]

OK: (https://github.com/Fabrizio-Caruso/Tiny-RPG/blob/main/src/test/test.c)
[image]

suborb, 2020-11-09:
The default for the stack is -1, see this thread: https://github.com/z88dk/z88dk/issues/592#issuecomment-363761086 

It doesn't look like you're using that much stack, but...it does look like you're using the firmware print implementation? I wonder if that has even more restrictions on register usage than we've previously thought.

Fabrizio-Caruso, 2020-11-09:
The problem is independent of scroll up. It just happens with a single printf but in my simple example it does bot happen. It does happen inif I also define some unused functions, unless I am missing something.
So it looks like the bigger binary somehow causes the issue.

suborb, 2020-11-09:
I've had a play around and it looks like the original problem is when compiled -clib=ansi.

It looks like the screen width is being set to 64 characters rather than 40 and that may well explain the behaviour (if you look how the wrapped characters are being printed it looks really strange).

The fix is quite small so you could apply it manually (it's just a config file change)

Fabrizio-Caruso, 2020-11-10:
Sorry for missing the clib option. Thanks for the fix!

closed 2020-11-10 by Fabrizio-Caruso