z88dk/z88dk #1639: help wanted on porting a program

zx70, 2020-11-11:
https://github.com/z88dk/z88dk-ext/blob/master/os-related/target.c

..port almost complete!  This is a nice piece of code for CP/M based on direct BDOS calls.
it works (a well emulated CP/M system is needed, I'm using uBee), but the interactive mode fails.  I think the reason is similar to the one preventing mm.c to work correctly, probably depending on the way  CR/LF is managed. 
Hints please?

zx70, 2020-11-11:
I forgot to mention that the original behaviour in command line mode didn't have the leading semicolon, but I noticed that such simple add fixed the problem.

zx70, 2020-11-14:
https://github.com/z88dk/z88dk-ext/blob/master/os-related/find.c
..in the meanwhile I could successfully convert this tool
It was originally written i Aztec C. While converting it I prepared a new header file,  aztecc.h to support future conversions.  It's still at a primitive stage though.

suborb, 2020-11-14:
Sorry, I thought you solved the problem, what's up with mm?

zx70, 2020-11-14:
That tool in interactive mode (add -DINTERACTIVE) does not detect the end of input loop

zx70, 2020-11-14:
if (!fgets(line,128, stdin))
	break;

suborb, 2020-11-15:
Try this:

```
--- a/libsrc/stdio/fgets_cons.c
+++ b/libsrc/stdio/fgets_cons.c
@@ -55,9 +55,9 @@ char *fgets_cons(char *str, size_t max)
            str[ptr] = 0;
            docursor();
         }
-      }
-      else
-      {
+      } else if ( c == 4 && ptr == 0 ) {
+         return NULL;
+      } else {
          if (_cons_state)
             c = toupper(c);
```

If Ctrl+D is the first character then we'll return. It's a fudge obviously since stdin isn't actually marked as EOF. That cold be done though.

zx70, 2020-11-16:
I tried the workaround, but no luck.   ^D has an effect, probably the expected one, but not enough to make the tool usable.    I'm now aware of a limit in the library, which is good.. I'll dig around both the program and the lib a bit more.

zx70, 2020-12-07:
For the moment I'm leaving the tool in a non interactive form.  The more interesting way to run it already works.

closed 2020-12-07 by zx70