z88dk/z88dk #1641: (sccz80) Separate backend and frontend

suborb, 2020-11-12:
sccz80 is derived from Small C so it builds up the output code as it parses.

If we separate parsing from code generation, then we have an opportunity to perform more optimisations beyond the limited constant folding, dead loop and assembler optimisations that we currently apply.

This will be a bit of a long-ish term subproject, so not planned for the next release unless there's another lockdown and the weather is awful!

I'm currently building the tree in parallel as the normal sccz80 operation so far we've got:

```
int main() {
        int k = 54;
        for ( int i = 0; i < 10; i++ ) {
                k = i * (10 + 20);
                func2(i);
        }
        return 22;
}
```

being turned into:

```
(decl k =54)
(decl i =0)
i_2:
(? (< (lv=i) 10) (noop) (goto i_3))
(= (lv=k) (* (lv=i) 30))
(func2 (lv=i))
(post++ (lv=i))
(goto i_2)
i_3:
(return 22)
```

Which already improves the shape of for loops, though I suspect I could tweak those slightly with sccz80 as it currently is.

spth, 2021-04-30:
Is it worth doing such a massive rework of sccz80?

Could contributing to sdcc be an alternative?

I wonder if it really makes sense for the community to maintain two separate free compilers for the z80, especially if both require major work, not just some minor maintenance.

suborb, 2021-05-01:
It's in a interesting series of questions and I can certainly see where you're coming from.

Is it worth it? I don't know: What's there works and the size/speed result after running through a few rules isn't too shabby considering the time taken to generate the result. However, this is something I've wanted to do for about 20 years so I'm happy to take my time and tinker with it when I get a chance.

Contributing to sdcc? I really don't have the time to do that, I don't even really have the time to do this issue (it's been dormant for 6 months after all), or really make significant progress on the other 193 issues here, so I don't think a bit of tinkering on sccz80 is going to be instead of any substantial contributions to sdcc. The time needed to implement features in sdcc needed to completely replace sccz80 (eg 8080/5 support, z80 port that doesn't use index registers, performance improvements) feels substantially larger than the time that I'm planning to spend working on sccz80.

suborb, 2021-05-20:
Obviously this isn't going to happen in any sort of sensible timescale, so closing.

closed 2021-05-20 by suborb