z88dk/z88dk #1642: Missing cat content when compiling for CP/M 8080

tsupplis, 2020-11-14:
Test program:
```
#include <stdio.h>

int main() {
    printf("hello");
    return 0;
}
```

Compilation:
```
zcc +cpm -m8080 -clib=8080 -lm -o base.com base.c
Error at file '__printf_number.asm' line 197: symbol '__math_block2' not defined
Errors in source file /opt/cc-z88dk/lib/config//../..//lib/target/cpm/classic/cpm_crt0.asm:
Error at file '__printf_number.asm' line 197: symbol '__math_block2' not defined
```

Workaround:
```
cp /opt/cc-z88dk/libsrc/z80_crt0s/8080/sccz80/crt_variables_math.asm math_vars.asm
zcc +cpm -m8080 -clib=8080 -lm -o base.com base.c math_vars.asm
```

Isn't crt_variables_math.asm supposed to be integrated into the crt? or is there a missing argument?

suborb, 2020-11-14:
It should be. 

I obviously messed it up when I thinned the libraries. I’ll sort it out. Apologies.

tsupplis, 2020-11-14:
Super thank you so much 😉...

suborb, 2020-11-20:
It looks like that issue was present back then, fixed, but then I introduced a slightly different issue a few weeks ago.

Regardless, it should all be fixed.

closed 2020-11-20 by suborb