z88dk/z88dk #1644: Bug in fread()

zx70, 2020-11-18:
When a multiple number of members is used on 'fread', something alters the position of the file after reading few members.

e.g. in "lar.c" (https://github.com/z88dk/z88dk-ext/blob/master/os-related/lar.c), a correct output would be:

-28JAN86.
XLT8-80.AQM
XLT8-80.COM
XLT8-80.DQC
TEST1B.MQC

.. but without the workaround below (in program line 1104), the output (with the commented code) will be:

-28JAN86.
XLT8-80.AQM
XLT8-80.COM
.
-28JAN86.
XLT8-80.AQM
XLT8-80.COM

--------  workaround currently in use --

//    if (fread ((char *) & ldir[1], DSIZE, nslots-1, f) != nslots-1)
//	error ("Can't read directory - is it a library?");

	// workaround for z88dk
	for (entry=1; entry<=nslots; entry++) {
		fread ((char *) & ldir[entry], DSIZE, 1, f);
	}

zx70, 2020-11-18:
I experimented a little more, and I get the same bad results also altering the original:
  fread ((char *) & ldir[1], DSIZE, nslots-1, f) 

to:
  fread ((char *) & ldir[1], DSIZE*(nslots-1), 1, f) 

Thus it can be a generic overflow problem in a single fread?
..or a limit in the file buffers?

suborb, 2020-11-18:
Is it hitting EOF and then rewinding? Is this at the fread() or read() level - my guess would be at read() level?

zx70, 2020-11-18:
Good and useful questions, thankyou!
The rewind() happens at the beginning, then a single fread() call would pick the whole archive directory, but it fails on z88dk (works with gcc on a modern PC).
Same with read(), which suggested me also a quick trick to provide fileno() to stdio.h

A loop on smaller chunks (DSIZE=32 bytes), fixes the problem.

zx70, 2020-11-18:
obviously this type of workaround would possibly fix other overflow conditions, I must pepare an alternative test case now

zx70, 2020-11-19:
ouch.. a smaller program including only the "failing code" works perfectly  :(

closed 2020-11-19 by zx70

zx70, 2020-11-21:
I finally tracked down the bug, the overflow was caused by a counter exceeding by a whole record (nslots needed a further decrement).   The side effect is something I still can't explain,  probably it was accidentally resetting the counter.
it's nice now to have fileno(), though.  Can I convert the macro into a real function?

suborb, 2020-11-21:
I honestly thought we had `fileno()` already! Sure, please add it, note that for std* it will return inappropriate numbers.

zx70, 2020-11-23:
..I improved the LAR tool quite a bit  ;)

C> lar e lt31.lbr
-README.1ST
LT31.FOR
-LT31FIL.LYT -> -LT31FIL.LST: lzh, done.
-LT31LBR.DYC -> -LT31LBR.DOC: lzh, done.
LT31.CYM -> LT31.COM: lzh, done.
LT31.DYC -> LT31.DOC: lzh, done.
LT31.MYC -> LT31.MAC: lzh, done.
UNC.RYL -> UNC.REL: lzh, done.
UNC.SYR -> UNC.SLR: lzh, done.
UNLZH.RYL -> UNLZH.REL: lzh, done.
UNLZH.SYR -> UNLZH.SLR: lzh, done.
LT31.COM
LT31.FQR -> LT31.FOR: sq, done.

C> lar e unzip15.lbr
UNZIP12.DZC -> UNZIP12.DOC: lzw, done.
UNZIP12.ZZ0 -> UNZIP12.Z80: lzw, done.
UNZIP15.CZM -> UNZIP15.COM: lzw, done.
UNZIP15.DZC -> UNZIP15.DOC: lzw, done.
UNZIP15.FOR
UNZIP15.ZZ0 -> UNZIP15.Z80: lzw, done.

zx70, 2020-11-25:
Ok, we have fileno() already, but it is missing in the classic lib.   I'm adding it now.

zx70, 2020-11-25:
..the bug in lar.c is still in!   the workaround mitigates it and the problem now visible anymore (without the workaround it is now visible only working on libraries with a long directory).
The tool does not seem to work on CP/M running on the Microbee emulator, though.

zx70, 2021-04-08:
With the issue #1745 I'm  hitting this (or a similar) bug again.  Apparently the new bug happens in read().