z88dk/z88dk #1645: git issue...

dikdom, 2020-11-20:
Hello,
I'm a re-starter here, I tried to sync my repo and build the latest z88dk package but my attempts failed..
So, here is what I did: I checked out my repo. Build the project successfully.
Then I added the upstream, fetched the upstream, merged the master branch, reset the environment, then executed build.sh -C. And it fails.
The order of my commands:

```
git clone  --recursive  https://github.com/dikdom/z88dk.git
cd z88dk
export BUILD_SDCC=1
./build.sh
; so far good, this one builds successfully. This is an old repo with an old state of Z88DK.
git remote add upstream https://github.com/z88dk/z88dk.git
git fetch upstream
git checkout master
git merge upstream/master
./set_environment.sh
time ./build -C
```
and this one fails after a couple of minutes with the following error:

```
-- Building z180 optimised crt routines---

make[1]: Nothing to be done for 'all'.
TYPE=z180 z88dk-z80asm -d -I/home/sanyi/devel/z88dk/lib/config//../ -mz180 -DSTANDARDESCAPECHARS -x/home/sanyi/devel/z88dk/libsrc//z180_opt @./z80_crt0s/crt0.lst
make: Leaving directory '/home/sanyi/devel/z88dk/libsrc'
make: Entering directory '/home/sanyi/devel/z88dk/libsrc'
cp /home/sanyi/devel/z88dk/libsrc//*.lib ../lib/clibs
make: Leaving directory '/home/sanyi/devel/z88dk/libsrc'
make: Entering directory '/home/sanyi/devel/z88dk/libsrc/_DEVELOPMENT'
m4 -DCFG_ASM_DEF target/am9511/config.m4 > target/am9511/config_am9511_private.inc
m4 -DCFG_ASM_PUB target/am9511/config.m4 > target/am9511/config_am9511_public.inc
m4 -DCFG_C_DEF target/am9511/config.m4 > target/am9511/config_am9511.h
cp target/am9511/config_am9511_private.inc target/am9511/obj/config_private.inc
z88dk-z80asm  -d -Otarget/am9511/obj/sccz80 -Itarget/am9511/obj -xlib/sccz80/am9511 -I/home/sanyi/devel/z88dk/libsrc/_DEVELOPMENT//../../lib/ @target/am9511/library/am9511_sccz80.lst
z88dk-z80asm  -d -Otarget/am9511/obj/sdcc_ix -Itarget/am9511/obj -xlib/sdcc_ix/am9511 -I/home/sanyi/devel/z88dk/libsrc/_DEVELOPMENT//../../lib/ -D__SDCC -D__SDCC_IX @target/am9511/library/am9511_sdcc_ix.lst
z88dk-z80asm  -d -Otarget/am9511/obj/sdcc_iy -Itarget/am9511/obj -IXIY -xlib/sdcc_iy/am9511 -I/home/sanyi/devel/z88dk/libsrc/_DEVELOPMENT//../../lib/ -D__SDCC -D__SDCC_IY @target/am9511/library/am9511_sdcc_iy.lst
m4 -DCFG_ASM_DEF target/cpm/config.m4 > target/cpm/config_cpm_private.inc
m4 -DCFG_ASM_PUB target/cpm/config.m4 > target/cpm/config_cpm_public.inc
m4 -DCFG_C_DEF target/cpm/config.m4 > target/cpm/config_cpm.h
cp target/cpm/config_cpm_private.inc target/cpm/obj/config_private.inc
z88dk-z80asm  -d -Otarget/cpm/obj/sccz80 -Itarget/cpm/obj -xlib/sccz80/cpm -I/home/sanyi/devel/z88dk/libsrc/_DEVELOPMENT//../../lib/ @target/cpm/library/cpm_sccz80.lst
Error at file 'target/cpm/driver/terminal/cpm_00_input_cons/cpm_00_input_cons_ichar_msg_getc.asm' line 61: symbol '__CPM_RCOB' not defined
Error at file 'target/cpm/driver/terminal/cpm_01_input_kbd_dcio/cpm_01_input_kbd_dcio_iterm_msg_getc.asm' line 20: symbol '__CPM_DCIO' not defined
Error at file 'target/cpm/driver/terminal/cpm_01_output_dcio/cpm_01_output_dcio_iterm_msg_bs.asm' line 18: symbol '__CPM_DCIO' not defined
Error at file 'target/cpm/driver/terminal/cpm_01_output_dcio/cpm_01_output_dcio_iterm_msg_bs.asm' line 24: symbol '__CPM_DCIO' not defined
Error at file 'target/cpm/driver/terminal/cpm_01_output_dcio/cpm_01_output_dcio_oterm_msg_bell.asm' line 21: symbol '__CPM_DCIO' not defined
Error at file 'target/cpm/driver/terminal/cpm_01_output_dcio/cpm_01_output_dcio_oterm_msg_putc.asm' line 67: symbol '__CPM_DCIO' not defined
Error at file 'target/cpm/driver/character/cpm_00_input_reader/cpm_00_input_reader_ichar_msg_getc.asm' line 20: symbol '__CPM_RRDR' not defined
make: *** [Makefile:120: lib/sccz80/cpm.lib] Error 1
make: Leaving directory '/home/sanyi/devel/z88dk/libsrc/_DEVELOPMENT'

real    28m26,435s
user    22m6,775s
sys     4m48,825s

```

I know I make a mistake during updating my local repo, because when I checkout the z88dk repo directly and build it, it succeeds.  Can you help me what do I miss here? Do I need to merge the sub-projects also, one by one? How?

Thanks for your support!

suborb, 2020-11-20:
I think you may have an old config_private.inc hanging around in libsrc/_DEVELOPMENT/config_private.inc

The location of this generated file was moved a couple of months ago, so the clean rule got removed.

If that is the problem, I'll restore the clean rule so that no one else runs into the problem.

dikdom, 2020-11-21:
Hello,
 
I removed the aforementioned ```libsrc/_DEVELOPMENT/config_private.inc``` file and that solved the problem. Thanks a lot for your support!

closed 2020-11-21 by dikdom

suborb, 2020-11-21:
@dikdom thanks for confirming the solution!