z88dk/z88dk #1649: coroutines - cooperative multithreading

feilipu, 2020-11-29:
I'm quite interested to build some kind of multithreading capability, that is not tied to having a system tick. Specifically, some targets (eg RC2014) don't inherently have a system tick capability, yet could benefit from having multi-threading.

In classic there is the [preemptive](https://github.com/z88dk/z88dk/tree/master/libsrc/threading/preempt) code (in C) and in newlib there is the [threads](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/threads) code. But both of these rely on the existance of a system tick.

An option to build this capability may be based on [Protothreads](http://dunkels.com/adam/pt/) (Adam Dunkels) in C (using Duff's Device), but perhaps the critical code could be migrated to assembly as needed. Lightweight, no stack or local variables.

Another alternative is to corrupt the [FreeRTOS Co-routine](https://www.freertos.org/croutineapi.html) code, and run it without the underlying RTOS timing elements. Heavyweight, with full context swap and stack.

Welcome thoughts or interest on which way would be the best or most generally useful?

__Outcome desired.__

Being able to separate an application into separate threads, and have them cooperatively "yield" when they're waiting on some long running event or code. My specific example is to code an application in C for a multi-APU capable RC2014. Up to 4 (or more) threads could each use their own APU to parallelise and break a problem into smaller pieces.

Eg. [RC2014 with 4 APU](https://twitter.com/_feilipu/status/1332924720258240514)

This could be done in assembly, but I feel it would be easier and more general to build the co-routine code once as a library, and then be able to build any further application in C.

feilipu, 2020-11-29:
Another obvious alternative is to use existing FreeRTOS preemptive code, without a system tick. I've already done this Z180 port, so simplifying is an option. This would probably be the most complete solution available.

Fabrizio-Caruso, 2020-12-26:
Cooperative coroutines in its lightest form (protothreads) could be implemented in a generic way for all targets. It is just a bunch of macros that do the "magic". They could have some sort of fake local variables (each new instance of a coroutines should have different local variables). 
Protothreads are extremely light.

feilipu, 2020-12-28:
IIRC someone here was playing with protothreads (Dunkel's version). I also had a go, but I don't really understand the "magic" well enough to debug.

What I'm actively thinking about now is trying to do a FreeRTOS without a system tick. Essentially each time the scheduler is called the system tick gets pre-incremented by one. That makes delays unworkable, but should allow the use of debugged supporting scheduling, queues and list code to a great extent. But, currently I'm diverted into moving into the 10.4.3 LTS code, and remediating issues around that on other platforms.

obiwanjacobi, 2021-01-18:
This is what I've got and works as is (Async): 
https://github.com/obiwanjacobi/Zalt/blob/4bedb0c2586469d834f9ceca2126d8141017f5cf/Source/z88dk_target/src/sys/Sys.h#L44
Example use:
https://github.com/obiwanjacobi/Zalt/blob/master/Source/z88dk_target/_tests.unit/sys/AsyncTests.c

It works on a trick of jumping in and out of a switch statement that implements the state machine.

Feel free to use.
[2c]

feilipu, 2022-01-05:
Working on other things. Will reopen if/when needed.

closed 2022-01-05 by feilipu