z88dk/z88dk #1652: SOS library: missing objects?

zx70, 2020-12-03:
@suborb,  what is the point of excluding the SOS variables position during the library build?
https://github.com/z88dk/z88dk/blob/master/include/sos.h

The resulting libraries claaim for missing objects during the link process.
[processor-core-factor-table-070634.pdf](https://github.com/z88dk/z88dk/files/5639199/processor-core-factor-table-070634.pdf)

suborb, 2020-12-03:
I'm not sure, I'm guessing it was so that the user could override them at compile time. However, the behaviour of that placement syntax has changed over the years. I have a feeling it's now file local by default?

zx70, 2020-12-04:
I moved the exports here:  https://github.com/z88dk/z88dk/blob/master/libsrc/target/sos/sos/sos_variables.asm
I think it is consistent also when the sdcc compiler is involved

closed 2020-12-05 by zx70