z88dk/z88dk #1654: [classic] Does AMALLOC still work on CP/M ?

zx70, 2020-12-08:
I tried to build the 'tail.c' example, which I put in the examples/osca folder some time ago.
"malloc.lib" is not any longer necessary since quite a bit now, but I appear to have included it in the build comments, so the bug (if ever) could be an old one.

The output is totally silent, no trailing lines are shown.   I'm suspecting the AMALLOC initialization, but I may be wrong.
Perhaps @suborb can share another bit of his kind wisdom and help in finding the failing element ?    :)

suborb, 2020-12-18:
I did get some output with +test, but I'm not convinced it was correct.

I'll put it on my todo list!

suborb, 2021-02-06:
I'm pretty certain that AMALLOC is setup correctly - modifying tail.c to read from a file rather than stdin shows the expected result.

Using a piped stdin just gives seems to skip characters with zxcc - I've not yet found the right zxcc command line magic to get the CP/M program to do the redirect.

Given that you said the redirection is working (so it's effectively reading from a file), I can't immediately see any reason why it wouldn't work!

zx70, 2021-02-06:
it is good news,   I'll repeat my tests with cp/m on the recent/clean daily.

Ottieni Outlook per Android<https://aka.ms/ghei36>

zx70, 2021-02-12:
I double checked, and my impression is that something is not still right (at least in 'classic' mode).
Sparse notes on the 'sort.c' and 'tail.c' programs in the examples/osca folder:

- First of all, to use them interactively, I need to type ^D.  I think it is due to the TEXT/BINARY way to build/setup the library, still fine as long as it is clear that ^Z won't act as an EOF

- The file redirection seems to work only when both the 'directions' are included, e.g. "sort <filein > fileout", if I keep 'filein' only, the command line parser gets confused (e.g. tail complains for a missing number of lines, also when specified)

- 'sort' is not sorting at all, I suspect that the EOL markers are being misinterpreted

- 'tail', if both the redirection parameters are passed, works as expected.

I tried to switch between different emulated environments during my checks, but I wasn't too methodic regarding this.

zx70, 2021-02-23:
I tried again with "osca/sort.c" ad "osca/tail.c" and the file redirection works, I was probably messing with my local shell file redirection options.
There are still 2 weird things: 
-when feeding data via stdin, I need to press ^D to close the file (I'd have expected ^Z ?)
-I tried on CP/M only, but the output file is never opened in "append" mode, again it can be a limit of the emulator I'm using

suborb, 2021-02-23:
The ^D is here: https://github.com/z88dk/z88dk/blob/master/libsrc/stdio/fgets_cons.c#L59

zx70, 2021-03-05:
The APPEND problem was in the FCNTL driver, not easy to spot.
It had 2 concurrent root causes, my messy implementation of lseek() and a wrong assignment of the RDWR flag, making read() fail,

zx70, 2021-03-05:
I'm so happy I found and fixed the problem! 
I could have been the responsible of the mess too..  :(

closed 2021-03-05 by zx70