z88dk/z88dk #1656: [info][bug?] Division with litterals

Fabrizio-Caruso, 2020-12-14:
I am trying to understand what goes on when I compile my code with an expression such as:
XSize-XSize/4-(XSize/5)

This seems to behave as I expect on all compilers including GCC but it seems I am having a problem with Z88DK compilers.

In my case XSize is 32 and I can even replace XSize with 32 and I still get something wrong as if 32/5 were not evaluated as integer division. i.e., 32/5 = 6. The result I get looks a bit like 32/5 = 7.

If I replace XSize/4 with 8 and XSize/5 with 6, I get the right expected result under Z88DK.

I am not sure this is a bug.

If this looks like a bug I can try to reproduce it in a simple example.

suborb, 2020-12-14:
Interesting. 

The expression folding is performed with floating point arithmetic and then converted to an int with a cast. Given the result is 17.6 it looks like we're truncating rather than rounding.

Fabrizio-Caruso, 2020-12-14:
@suborb should the division between integer litterals rather be integer division?

Fabrizio-Caruso, 2020-12-14:
@suborb It looks like GCC uses integer division to perform division between integer litterals (as normal C code)
    printf("debug 2: %d\n", XSize/5); -> 6
    printf("debug 2: %f\n", XSize/5); -> non-sense

Fabrizio-Caruso, 2020-12-14:
@suborb If we assume that GCC (and all other compilers I have tried are correct), then the problem is not the rounding. 
The problem is that the wrong operation is performed on integer litterals.

Fabrizio-Caruso, 2020-12-14:
If you apply the rounding and keep float operations on integer litterals, we will get errors elsewhere, i.e., when float fractions add up to an integer.

suborb, 2020-12-14:
Yup, It's obviously not right and it'll propagate through all calculations.

I'll fix it, though I'm having a bit of time "away" from z88dk at the moment so it'll be a few days.

Fabrizio-Caruso, 2020-12-17:
The problem seems fixed for me.

closed 2020-12-17 by Fabrizio-Caruso