z88dk/z88dk #1657: [ZSDCC only] extern definition mismatches with declaration

Fabrizio-Caruso, 2020-12-17:
It seems that ZSDCC (but not SCCZ80) cannot tollerate extern arrays coupled with arrays definitions without an explicit size, i.e.,
`const uint8_t level_walls[] =  `
will fail when coupled with
`const uint8_t level_walls[]`
and produce something like:
```
games/snake/main.c:253: error 91: extern definition for 'level_walls' mismatches with declaration.
games/snake/levels.h:5: error 177: previously defined here
```

Is this normal correct behavior?

suborb, 2020-12-21:
Have you got an example/which of your repos is this in?

Fabrizio-Caruso, 2020-12-21:
@suborb I will try to create a simple test case. I am not sure it is a bug as SDCC may be more compliant with the ANSI standard. SCCZ80 tollerates definitions with declarations with unsized arrays whereas ZSDCC seems to complain.

I have the problem on the Z88DK targets in:
https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/src/Makefile.snake
(some cannot be compiled with ZSDCC for other reasons).

Fabrizio-Caruso, 2020-12-23:
I am closing this. Sorry. I am not longer able to reproduce it...

closed 2020-12-23 by Fabrizio-Caruso