z88dk/z88dk #1659: Full z88dk Installation with SDCC Installation on Ubuntu Linux 20.04

spectrumcomputing, 2020-12-21:
Hi,

I'm installing z88dk on Ubuntu 20.04.

I install all the required packages:

    sudo apt-get install dos2unix libboost-all-dev texinfo texi2html libxml2-dev subversion bison flex zlib1g-dev

I download the nightly build:

    wget http://nightly.z88dk.org/z88dk-latest.tgz
    tar -xzf z88dk-latest.tgz

Then follow the instructions:

    cd z88dk
    export BUILD_SDCC=1
    chmod 777 build.sh
    ./build.sh

When the installer gets to this line, it just hangs:

    svn checkout -r 11940 https://svn.code.sf.net/p/sdcc/code/trunk/sdcc -q /home/ubuntu/z88dk/src/sdcc-build 

I have checked that this site is available:

https://svn.code.sf.net/p/sdcc/code/trunk/sdcc/

Many thanks

Peter JOnes

suborb, 2020-12-21:
Hi, sdcc's SVN can be horrendously slow at times, so I ended up adding a "download tarball" option for the CI systems.

You can use it too, if you additionally add `export BUILD_SDCC_HTTP=1` before the `./build/sh` line

spectrumcomputing, 2020-12-21:
Hi @suborb,

Thanks for the quick reply.  Results!  Thank you.  Is there any chance you could add this to the install instructions, as I struggled for a number of days, before raising a bug.  Best wishes and festive greetings.  Peter Jones

suborb, 2020-12-21:
No worries, it's already on the wiki page: https://github.com/z88dk/z88dk/wiki/installation - but it may be a bit hidden:

> This will build z88dk include zsdcc. If you don't want to build zsdcc then omit the export BUILD_SDCC=1 line. The source code for sdcc is downloaded from svn which can be slow, if it keeps failing then run the command: export BUILD_SDCC_HTTP=1 before prior to running ./build.sh - this will download a source code tarball from the z88dk nightly website.

spectrumcomputing, 2020-12-21:
Thanks, can I ask one more question?  This did not seem to include appmake.  Do I need to install that separately?

suborb, 2020-12-21:
No, it's included. There were a lot of clashes with other packages with the name of the z80asm binary, so for consistency most of the tools got a z88dk- prefix a few months ago.

spectrumcomputing, 2020-12-21:
Thank you! Got it.

closed 2020-12-21 by spectrumcomputing