z88dk/z88dk #1661: [bug?] CONIO (gotoxy) seems to be broken in MSX

Fabrizio-Caruso, 2020-12-30:
I am not able to use mode 2 (color bitmap mode with color attributes that should be used with UDG and CONIO)

I use:
`
#include <graphics.h>
#include <sys/ioctl.h>
#include <conio.h>
int main(void)
{
    int mode = mode_2;
    console_ioctl(IOCTL_GENCON_SET_MODE, &mode);
    gotoxy(2,2);
    printf("hello world!\n");
    gotoxy(0,0);
    printf("a");
    gotoxy(32,0);
    printf("b");
    gotoxy(24,0);
    printf("c"); 
    gotoxy(32,24);
    printf("d");
    while(1){};
    return 0;
}
`
which I compile with:
`
msx_mode2_test:
	$(Z88DK_PATH)$(MYZ88DK) +msx $(SCCZ80_OPTS) -subtype=rom \
	-pragma-redirect:fputc_cons=fputc_cons_generic -vn -lndos \
	$(TEST_PATH)//msx_test2.c \
	-create-app -o $(BUILD_PATH)/TEST_msx2.prg
	rm $(BUILD_PATH)/TEST_msx2.prg
`

I get this:
[image]

suborb, 2020-12-30:
Isn't gotoxy() 0 index based? - Your code looks like it's using a mix of 0 and 1 based indexing.

So when you go out of bounds it will scroll - hence why you don't see some of the letters.

Fabrizio-Caruso, 2020-12-31:
@suborb you are right (as usual)... I am not able to get my game to work in mode 2 for some other reason... I need to investigate this better... Sorry for this issue.

closed 2020-12-31 by Fabrizio-Caruso

suborb, 2020-12-31:
That’s odd. I’m guessing it’s the same code as the other vdp targets? The display library really doesn’t care what machine you’re running on.

Fabrizio-Caruso, 2020-12-31:
What I do with the other UDG and CONIO enabled targets is:
```

#include <graphics.h>

#include "display_macros.h"

#include <sys/ioctl.h>

extern uint8_t udgs[];

void INIT_GRAPHICS(void)
{
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);
	
	#if defined(FORCE_SCREEN_MODE) || defined(__MC1000__) || defined(__PC6001__) || defined(__CPC__) \
    || defined(__MULTI8__) || defined(__SPC1000__) || defined(__GB__) || defined(__SMS__) \
    || defined(__GAL_PLUS__) || defined(__GAME_GEAR__) || defined(__MTX__) || defined(__MSX__)
	{
		#if defined(FORCE_SCREEN_MODE)
			#define SCREEN_MODE FORCE_SCREEN_MODE
		#elif defined(__MC1000__) || defined(__GB__) || defined(__GAL_PLUS__) 
			#define SCREEN_MODE 1
		#else
			#define SCREEN_MODE 2
		#endif
		int mode = SCREEN_MODE;
		console_ioctl(IOCTL_GENCON_SET_MODE, &mode);
        
	}
	#elif defined(__SPECTRUM__)
	printf("\x01\x20");
    zx_border(0);
    textbackground(COLOR_BLACK);
	#endif
}
```
which, for some reason, does not work on the MSX which gives me some corrupted graphics.

Fabrizio-Caruso, 2020-12-31:
which I compile with:
```
msx_test: msx_32k_rom_2

msx_32k_rom_2: $(ASSETS_PATH)/z88dk_xchase.asm  
	$(Z88DK_PATH)$(MYZ88DK) +msx $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
	-pragma-redirect:fputc_cons=fputc_cons_generic \
	-DINITIAL_SLOWDOWN=5000 \
	-DSOUNDS -DANIMATE_PLAYER \
	-vn  -DFULL_GAME -D__MSX__ -DEND_SCREEN -DBETWEEN_LEVEL \
	-lndos -subtype=rom \
	-create-app -o $(BUILD_PATH)/X$(GAME_NAME)_msx_32k.rom \
	-DCONIO -DFORCE_SCREEN_MODE=2 -DFORCE_NARROW -DFORCE_XSIZE=32 -DFORCE_YSIZE=24 \
	-DUDG_GRAPHICS \
	-DZ88DK_JOYSTICK \
	-DSLOW_DOWN=2700 \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(CROSS_LIB_PATH)/sound/z88dk/psg/psg_sounds.c \
	$(FULL_FILES)
	rm $(BUILD_PATH)/X$(GAME_NAME)_msx_32k_BSS.bin
	rm $(BUILD_PATH)/X$(GAME_NAME)_msx_32k_DATA.bin
```

Fabrizio-Caruso, 2020-12-31:
In case you want to give it a try:
```
make msx_test -f Makefile.snake
```
builds a version of Cross Snake that should use CONIO and UDG but has glitches.
while
```
make msx -f Makefile.snake
```
builds Cross Snake with my own direct video memory writes in mode 1 and it works fine.

I cannot exclude some stupid bug on my own code. Sorry if this is the case. I need to investigate this better.