z88dk/z88dk #1662: Both zsdcc and sccz80 both miscompile code with static variables

imneme, 2021-01-02:
I wrote a C implementation of a solution for an _Advent of Code_ puzzle (Day 19) to run under CP/M (under both `tnylpo` and on a _Spectrum Next_).

In an effort to reduce execution time due to costly indexing operations on the stack, since the functions were non-reentrant, I switched the code to use function-static variables.  The code runs very nicely using Hi-Tech C (and Clang, including checking with its various sanitizers), but both sccz80 and zsdcc miscompile the code.

Here's a session:
```
ubuntu@ubuntu:~/spectrum$ z88dk.zcc +cpm -DAVOID_SCCZ80_BUG regexp-s.c -o zdrexp-s.com  && tnylpo zdrexp-s
Matched (correct)

ubuntu@ubuntu:~/spectrum$ z88dk.zcc +cpm regexp-s.c -o zdrexp-s.com  && tnylpo zdrexp-s
Assertion failed: regexp-s.c 71 "ruleno >= 0 && ruleno < RULES_MAX"

ubuntu@ubuntu:~/spectrum$ z88dk.zcc +cpm -compiler=sdcc -DAVOID_SDCC_BUG regexp-s.c -o zdrexp-s.com  && tnylpo zdrexp-s
Matched (correct)

ubuntu@ubuntu:~/spectrum$ z88dk.zcc +cpm -compiler=sdcc regexp-s.c -o zdrexp-s.com  && tnylpo zdrexp-s
tnylpo: HALT instruction executed
```

Similar results apply when using newlib, optimization options, `--reserve-regs=iy`, etc.

Here's the code.  I've whittled it down, but perhaps not as much as you'd like. Sorry if the algorithm isn't the easiest to fathom.

```
#include <stdio.h>
#include <stddef.h>
#include <stdarg.h>
#include <assert.h>

enum { RULES_MAX = 4 };

typedef int* rule_seq;
typedef rule_seq* rule_alts;

int       rule_seq_lists[] = { 3, 1, 3, -1,
			       2, -1,
			       2, 1, -1 };
rule_seq  rule_alts_lists[] = { &rule_seq_lists[0], NULL,
				&rule_seq_lists[4], &rule_seq_lists[6], NULL };
rule_alts rule[RULES_MAX] = { &rule_alts_lists[0],
			      &rule_alts_lists[2],
			      NULL,
			      NULL };
char      rule_terminal[RULES_MAX] = { 0, 0, 'a', 'b' };

enum { GOOD_POOL_MAX = 64,
       BAD_STACK_MAX  = 16 };

struct good {
   rule_seq seq;
   struct good* prev;
} good_pool[GOOD_POOL_MAX+1];  /* +1 to avoid incorrect SDCC warning */

struct bad {
   rule_alts    alts;
   char*        old_pos;
   struct good* old_good;
} bad_stack[BAD_STACK_MAX+1];  /* +1 to avoid incorrect SDCC warning */

int match(char* pos_arg)
{
    static char* pos;
    static struct bad*  bad_sp;
#ifdef AVOID_SCCZ80_BUG
    #define static
#endif
    static struct good* good_alloc;
    static struct good* good;
    static struct good* rgood;
    static rule_alts    alts;
#ifdef AVOID_SDCC_BUG
           rule_seq     seq;
#else
    static rule_seq     seq; /* this line breaks SDCC */
#endif
#ifdef AVOID_SCCZ80_BUG
    #undef static
#endif
    static char*        rpos;
    static int          ruleno;
    pos        = pos_arg;
    bad_sp     = bad_stack;
    good_alloc = good_pool;
    good       = NULL;
    rgood      = NULL;
    alts       = NULL;
    seq        = NULL;
    rpos       = NULL;
    ruleno     = 0;
    
    while (1) {
        assert(bad_sp < &bad_stack[BAD_STACK_MAX]);
        assert(good_alloc < &good_pool[GOOD_POOL_MAX]);
        assert(ruleno >= 0 && ruleno < RULES_MAX);
        alts = rule[ruleno];
        if (alts) {
            rgood = good;
            rpos = pos;
            goto first_alt;
        }
        assert(rule_terminal[ruleno]);        
        if (*pos != rule_terminal[ruleno])
            goto failure;
        ++pos;
    success:
        if (!good) {
            if (*pos != '\0')
                goto failure;
            return 1;
        }
        seq = good->seq;
        good = good->prev;
        ruleno = *seq++;
        if (*seq >= 0) {
            struct good* ngood = good_alloc++;
            ngood->seq = seq;
            ngood->prev = good;
            good = ngood;
        }
        continue;
    failure:
        if (bad_sp == bad_stack) {
            return 0;       
        }
        --bad_sp;
        alts = bad_sp->alts;
        rgood = bad_sp->old_good;
        rpos = bad_sp->old_pos;
    first_alt:
        good = good_alloc++;
        good->seq = *alts++;
        good->prev = rgood;
        pos = rpos;
        if (*alts) {
            bad_sp->alts = alts;
            bad_sp->old_good = rgood;
            bad_sp->old_pos  = pos;
            ++bad_sp;
        }
        goto success;
    }
}

int main()
{
    if (match("baaab")) {
        printf("Matched (correct)\n");
    } else {
        printf("Failed to match (incorrect)\n");
    }
    return 0;
}
```

imneme, 2021-01-02:
Oh, I should add that this is with the current version on snap. Namely,
```
Small-C/Plus - z80 Crosscompiler   Version: 17194-4fa760eb0-20201228
```
and
```
ZSDCC IS A MODIFICATION OF SDCC FOR Z88DK
Build: 4.0.4 #11935 (Linux) Dec 28 2020
```
The same issues appear with version from a few months ago, so it's not a recent regression.

suborb, 2021-01-02:
The sccz80 was a long standing bug which hasn't been spotted for a few years rather weirdly.

The sdcc one I've not investigated lately, but we're probably due an sdcc upgrade at some point soon - and I know a lot of changes have taken place there.

feilipu, 2021-01-02:
If possible, please post upstream on the SDCC bug tracker.

https://sourceforge.net/p/sdcc/bugs/

The team like to have their issues there. They also have multiple architectures to worry about.

Thanks. Will have a look too, and add to your SDCC issue ticket.

imneme, 2021-01-03:
Okay. Reported [here](https://sourceforge.net/p/sdcc/bugs/3166/).

imneme, 2021-01-03:
> The sccz80 was a long standing bug which hasn't been spotted for a few years rather weirdly.

I can confirm that retesting with 17199-a5539b895-20210102 shows the issue is fixed in sccz80. Awesome! Thanks!

> The sdcc one I've not investigated yet, but we're probably due an sdcc upgrade at some point soon - and I know a lot of changes have taken place there.

I checked with 4.0.7 #12003, which seems to be the current snapshot, and the miscompilation persists there.

feilipu, 2021-01-30:
Possibly also addressed by #1691 ?

imneme, 2021-01-31:
Still broken with the current edge snap:
```
ubuntu@ubuntu:~/spectrum$ z88dk.zcc -v
A config file must be specified with +file as the first argument

zcc - Frontend for the z88dk Cross-C Compiler - v17412-0d0e48c87-20210130
[...]

ubuntu@ubuntu:~/spectrum$ /snap/z88dk/current/bin/zsdcc -v

ZSDCC IS A MODIFICATION OF SDCC FOR Z88DK
Build: 4.0.7 #12036 (Linux) Jan 30 2021
[...]
```

```
ubuntu@ubuntu:~/spectrum$ z88dk.zcc +cpm -compiler=sdcc regexp-s.c -o zdrexp-s.com
ubuntu@ubuntu:~/spectrum $ tnylpo zdrexp-s.com
tnylpo: HALT instruction executed
```

imneme, 2021-07-04:
This bug is now [fixed upstream](https://sourceforge.net/p/sdcc/code/12527/).

feilipu, 2021-08-23:
Closed by #1838.

suborb, 2021-09-23:
#1838 was merged so this one is now closed.

closed 2021-09-23 by suborb