z88dk/z88dk #1663: [question/bug][cpc] Colors for border and background

Fabrizio-Caruso, 2021-01-02:
Is it possible to set the border color on the CPC target through bordercolor(c)?

Is setting the background text color before a clear screen, supposed to produced a black background?

I am using:
```
#include <conio.h>
#include <graphics.h>
#include <sys/ioctl.h>
int main(void)
{
    int mode = 0;
    console_ioctl(IOCTL_GENCON_SET_MODE, &mode);   
    textbackground(BLACK);
    bordercolor(BLACK);
    clrscr(); 
    textcolor(RED);
    gotoxy(2,2);
    printf("Hello world\n");
    while(1){};
    return 0;
}
```
which I compile with
```
cpc_test:
	$(Z88DK_PATH)$(MYZ88DK) +cpc -lndos -pragma-define:REGISTER_SP=-1 -clib=ansi  -create-app -subtype=dsk -o$(BUILD_PATH)/COLOR  -vn \
	$(TEST_PATH)/cpc_test.c
```
and the run it with:
```
memory &11ff
load "color.cpc",&1200
call &1200
```
The result I see on JavaCPC (which should be an accurate emulator):
[image]

suborb, 2021-01-02:
It looks like the colour mapping for the CPC is wrong - I have a feeling I wasn't too sure how to handle mapping between mode 0 and mode 1 colours so I didn't really do it properly.

Fabrizio-Caruso, 2021-01-02:
@suborb, It looks like there is also a problem with `textbackground` as it works on text but it is lost after `clrscr.` Am I wrongly assuming that `clrscr` should apply `textbackground` to the all screen?

suborb, 2021-01-02:
I think it should now work correctly? The palette is redefined on screen mode change to match the ANSI colour list.

Hopefully that'll sort it out (or at the very least make it better!)

closed 2021-01-02 by suborb

Fabrizio-Caruso, 2021-01-02:
Thanks a lot!
Is `textbackground` supposed to check the background of printed text even after a `clrscr?`
It seems its effect is lost after `clrscr`.
I had assumed that setting textbackground to black or whatever color would also change the way clrscr works.

suborb, 2021-01-02:
It should do....

https://github.com/z88dk/z88dk/blob/master/libsrc/target/cpc/stdio/generic_console.asm#L55

Fabrizio-Caruso, 2021-01-02:
@suborb then we may have a probable independent second bug as I get this just before `clrscr:`
[image]

which shows that textbackground does work (even the right color but that is not the point).

But after the code reaches clrscr, then the second printf loses the background color previously set:
[image]

I have used the following code:
```
#include <conio.h>
#include <graphics.h>
#include <sys/ioctl.h>
int main(void)
{
    int mode = 0;
    console_ioctl(IOCTL_GENCON_SET_MODE, &mode);
    
    textbackground(BLACK);
    printf("Hello world\n");
    getchar();
    
    bordercolor(BLACK);
    clrscr(); 
    textcolor(RED);
    gotoxy(2,2);
    printf("Hello world\n");
    while(1){};
    return 0;
}
```

suborb, 2021-01-02:
I think I may have inadvertently fixed that then as well :)

[image]

[image]

Fabrizio-Caruso, 2021-01-03:
@suborb it now works fine with a minor problem: red (pencil 4 in BASIC) is mapped to pink (pencil 7 in BASIC). A fix may just be a matter of switching 7 with 4.

[image]

suborb, 2021-01-03:
That should be an easy fix - I just went off the names rather than the look, mind you I’m even worse when it comes to matching colours by eye!