z88dk/z88dk #1665: (classic)(svi) Request for ROM build config for SVI

james7780, 2021-01-03:
SVI has build config for .cas file and also disk file.
Is it possible to add a build config to make a 32k ROM file (as with MSX)?
(I have tried modifying local copy of svi.cfg and svi_crt0.asm, but no success).

suborb, 2021-01-03:
Sure - have you got a pointer to some docs for the rom header? Or is it identical to the msx rom? 

I‘ll have a read of the mame source code to see if megaroms are supported.

suborb, 2021-01-03:
I've just committed some code that will generate a 32k ROM for the SVI.

I've probably missed a few things, but it seems to work on Mame. There's no access to the firmware, so for keyboard input you'll have to add the --hardware-keyboard option.

Let me know how it goes...

closed 2021-01-03 by suborb

james7780, 2021-01-05:
Thank you very much - I will try it out and give feedback.

As far as I could work out, the SVI ROM cart has no header, but must begin with 0xF3 0x31  (ie: Z80 asm code for "DI" and "LD SP,XXXX"), for the machine to recognise it and bank-switch it in.  (I could be wrong).

james7780, 2021-01-06:
Feedback:
- I did manage to build my project to a 32k SVI ROM using the new config. Many thanks for that.
- Since the cartridge ROM is mapped over the BIOS/BASIC ROM, any BIOS call will probably crash the ROM program.
- With MSX, the cartridge ROM is not mapped over the BIOS/BASIC ROM, so the BIOS calls in the MSX/SVI library functions (eg: msx_color() ) will work.
- TL;DR - You cannot build an MSX ROM and an SVI ROM from the same code. The SVI ROM code must implement replacements for the MSX lib functions which call BIOS routines.
- However you can still build .cas / disk image from the same code.

suborb, 2021-01-06:
Yes, I think it's expected that BIOS routines won't work as it stands. I suppose an interposer that does the banking trampolines could be written..

However, -lsvibios doesn't add a massive amount of value (IMO). It's equally portable to use the z88dk abstraction layer:

* `--generic-console` and `<conio.h>` for text/UDG printing 
* `--hardware-keyboard` is a transparent drop-in for keyboard reading
* `joystick()` for reading joysticks

Not using -lsvibios/-lmsxbios means that your code is then portable across all VDP (and non-VDP) machines.