z88dk/z88dk #1669: [c128] Wrong bit bang sounds

Fabrizio-Caruso, 2021-01-13:
The following code can be used to test `bit_fx(7)` and hear its reproduction
```
#include <sound.h>
#include <stdio.h>
int main(void)
{
    while(1) {
        bit_fx(7);
        getchar();
    }   
    return 0;
}
```

I have attached the wrong sound effect as reproduced by the emulated C128 (Vice):

[2021-01-13 09-35-55.zip](https://github.com/z88dk/z88dk/files/5807391/2021-01-13.09-35-55.zip)

zx70, 2021-01-13:
Are you getting this same behavior also from different environments, such as other emulators and/or a real C128?
The way the bit-banging works on the SID was is not orthodox at all and perhaps the emulators are not expecting it.

Fabrizio-Caruso, 2021-01-13:
I have not used a readl C128 but I have used Z64K and Vice.
Z64K (despite its name) is the most accurate C128 emulator.

Z64K has the same (not identical sounds but somehow similar) problem as Vice.

Some sounds are correctly reproduced by both Vice and Z64K and this may give you a hint for a fix.
For example in the code below, the first two samples are totally wrong and the third one sounds OK.
```
#include <sound.h>
#include <stdio.h>
int main(void)
{
    while(1) {
        bit_fx(7);
        getchar();
        
        bit_fx(1);
        getchar();
        
        bit_fx3(0);
        getchar();
    }   
    return 0;
}
```

zx70, 2021-01-13:
It is a good clue, indeed.
The effects you are claiming not to play correctly are the ones depending on "beeper.asm".
It is caused by the sound initialization (and reset) code being called at every note variation in the fx code loop.

zx70, 2021-01-13:
It should be fixed now.

zx70, 2021-01-13:
Impact was on bit_fx() and bit_fx4() effect groups only.   Now solved.

Fabrizio-Caruso, 2021-01-14:
@zx70 The C128 target has good sounds now but your changes have broken something elsewhere.
On the Spectrum: the sounds will also change the border colors. I fear something really bad could happen on other targets.

Fabrizio-Caruso, 2021-01-14:
@zx70 you can reproduce the problem by using my test program above and compile it for the spectrum target.
The border color is set to purple when the previously wrong sounds are reproduced:
`bit_fx(7),` `bit_fx(1)` while `bit_fx3(0)` resets to its previous color.

zx70, 2021-01-15:
Good news, the past change was consistent with the Spectrum and other targets, the was just an extra small step missing.

zx70, 2021-01-15:
Solved !

closed 2021-01-15 by zx70

zx70, 2021-01-15:
Part of the work was useful also to close the issue #1667 related to the bit-banging sound routines messing up the graphics mode.