z88dk/z88dk #1670: Target CPM and code Intel 8080

sblendorio, 2021-01-15:
Hi, I compiled just an "Hello World":

    #include <stdio.h>
    
    void main(void) {
        printf("Vediamo se funziona\n");
    }

with this command line:

    z88dk +cpm -m8080 test.c -o test.com

but, disassembling the ".com", I found specific Z-80 code (like LD BC and registers IX and IY). Trying to execute it on CP/M running on 8080, it does not run and hangs up the machine.

This is the .com file resulting: https://www.dropbox.com/s/fehqtae0nvxv9wb/test1.com?dl=0

This is the test I did:

[image]

The .com file was also tested on a real Intel 8080 machine with CP/M.
How can I get a compiled .com for CP/M for Intel 8080?

suborb, 2021-01-15:
The 8080 target is specified using `-clib=8080`, so the command line should be:

```
zcc +cpm -clib=8080 test.c
```

sblendorio, 2021-01-15:
IT WORKED! Thanks a  lot!

closed 2021-01-15 by sblendorio