z88dk/z88dk #1673: Random error about asm_fpclassify but I do not use floating point numbers

Fabrizio-Caruso, 2021-01-17:
I am getting a random compilation error that I cannot reproduce as it happens when I compile multiple targets  (NOT in parallel):
```
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa__.asm' line 48: symbol 'asm_fpclassify' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa__.asm' line 73: symbol '__dtoa_base10' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa__.asm' line 106: symbol '__dtoa_digits' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa__.asm' line 140: symbol '__dtoa_digits' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa_preamble.asm' line 54: symbol '__dtoa_sgnabs' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa_print.asm' line 34: symbol '__stdio_printf_sign_0' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoe__.asm' line 47: symbol 'asm_fpclassify' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoe__.asm' line 72: symbol '__dtoa_base10' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoe__.asm' line 97: symbol '__dtoa_digits' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoe__.asm' line 105: symbol '__dtoa_digits' not defined
Error at file 'sccz80/crt0_z80/fa.asm' line 12: symbol 'init_floatpack' not defined
Errors in source file c:/z88dk/\lib\target\mc1000\classic\mc1000_crt0.asm:
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa__.asm' line 48: symbol 'asm_fpclassify' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa__.asm' line 73: symbol '__dtoa_base10' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa__.asm' line 106: symbol '__dtoa_digits' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa__.asm' line 140: symbol '__dtoa_digits' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa_preamble.asm' line 54: symbol '__dtoa_sgnabs' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa_print.asm' line 34: symbol '__stdio_printf_sign_0' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoe__.asm' line 47: symbol 'asm_fpclassify' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoe__.asm' line 72: symbol '__dtoa_base10' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoe__.asm' line 97: symbol '__dtoa_digits' not defined
Error at file '/home/build/z88dk/build/z88dk/libsrc/_DEVELOPMENT/stdlib/z80/__dtoe__.asm' line 105: symbol '__dtoa_digits' not defined
Error at file 'sccz80/crt0_z80/fa.asm' line 12: symbol 'init_floatpack' not defined
make: *** [Makefile.tiles:729: mc1000_exp_48k] Error 1
```

suborb, 2021-01-17:
I can't reproduce it on your repo so I assume it's code you're working on at the moment.

I can reproduce this if I do a `printf("%f")` without linking to a maths library - do you have an errant logging line somewhere?

Fabrizio-Caruso, 2021-01-18:
It is weird as it fails when I am mass-compiling but I do not see it when I compile just one target.
It may be related to printf("%f") as I am using CONIO to display user-defined characters and their codes may coincide with % and f in some cases.... 
I think I can disable formatters with pragmas.

Fabrizio-Caruso, 2021-01-25:
@suborb For example if you take the current master (70d6a79b2a0cff0134f1543eeb36b30e43148c66) of my project, you may be able to reproduce the problem by running `make all -f Makefile.tiles`.

But I am not sure at all you will get the error... as it is happening in an unpredictable way.

Fabrizio-Caruso, 2021-04-27:
@suborb I am still getting this problem and I don't have any clue...
I run `make all -f Makefile.tiles` and I get the error.
I rerun it just afterwards and it works fine (!?!),

I suspect some of generated temporary files.
I may be getting this problem only with Makefile.tiles.

Any idea or suggestion?

suborb, 2021-05-01:
I've never been able to reproduce this one.

I think I know how it could get included: if printf detects a %f in the format string then it will pull in the floating point converter which ends up depending on dtoa.

How it ends up detecting that is the mysterious bit - we could go on an investigation loop but it might be easier for you just to add a `#pragma printf "needed format values"` to a zpragma.inc: https://github.com/z88dk/z88dk/wiki/Classic--Pragmas and override the auto detection.

feilipu, 2022-02-17:
Is this still an issue?
Or could it be closed?

feilipu, 2022-10-02:
Assumed resolved.

closed 2022-10-02 by feilipu