z88dk/z88dk #1676: Bug in "strtol" from stdlib in Intel 8080 target

sblendorio, 2021-01-19:
Let's compile this program for CP/M, for both 8080 and Z80 CPUs:

    #include <stdio.h>
    #include <stdlib.h>
    
    void main(void) {
        char *t;
        printf("-- BEGIN --\n");
        printf("%ld\n",strtol("0x0a", &t, 16));
        printf("%ld\n",strtol("0a", &t, 16));
        printf("%ld\n",strtol("0A", &t, 16));
        printf("%ld\n",strtol("0X0a", &t, 16));
        printf("%ld\n",strtol("0x0A", &t, 16));
        printf("%ld\n",strtol("0X0A", &t, 16));
        printf("%ld\n",strtol("0x00A", &t, 16));
        printf("%ld\n",strtol("0X000A", &t, 16));
        printf("%ld\n",strtol("0x0000a", &t, 16));
        printf("-- END --\n");
    }

It just parses the "0a" hex number into a long (dec=10) and **prints "10" several times**. Compiling it with:

    $ zcc +cpm test.c -o test.com

...it works **as expected**. While, compiling it for **Intel 8080**, this way:

    $ zcc +cpm -clib=8080 test.c -o test.com

...it will produce a **broken executable**, repeating several time the first printf ("BEGIN") till (maybe) the stack is full:

    A>test
    -- BEGIN --
    -- BEGIN --
    -- BEGIN --
    -- BEGIN --
    -- BEGIN --
    -- BEGIN --
    -- BEGIN --
    -- BEGIN --
    -- BEGIN --
    ...

suborb, 2021-01-19:
As was predictable....there's tests for strtol() but the 8080 test case wasn't enabled.

I shall enable and fix - thanks for the report.

suborb, 2021-01-20:
Oh this is an odd one, it took me a while to reproduce it - the cause was the base was being left on the stack so it was returning to address 0x10.

I initially ran under ticks (+test target), but at address 0x10 there is effectively a ret so it wasn't noticed.

I then tried zxcc (+cpm -clib=8080), but it too must have a ret effectively at that address.

Final try was ticks in CP/M mode, here nothing is located at 0x10, so execution runs onto address 0x100 and the program starts again. I guess it eventually dies when the stack overwrites the program.

closed 2021-01-20 by suborb

sblendorio, 2021-01-21:
I did a **git pull**, a **make clean** and **make install**, I got this version number:

    zcc - Frontend for the z88dk Cross-C Compiler - v17267-67ad7aec2-20210114

is it the latest one wih this patch applied?

feilipu, 2021-01-21:
Try deleting the compiled contents of ~/z88dk/bin (but not zsdcc or zsdcpp). That helps before you build.

Also use `./build.sh` rather than `make`. The `./build.sh` script does more than just make.

sblendorio, 2021-01-21:
perfet. It worked. Thank you!