z88dk/z88dk #1677: Possible bug with zcc

vipoo, 2021-01-21:
I assume this is the place to log errors with the c compiler - specifically the sdcc variant...

Stumbled on an issue tonight with the c compiler producing incorrect z80 code.

I have tried to shrink it down to the simplest example to demonstrate the issue:

filename: spike.c
```
#include <stdbool.h>
#include <stdlib.h>

typedef struct _Struct {
  uint8_t abyte1;
  uint8_t abyte2;
  uint16_t aint;
} Struct;

Struct storage;

uint16_t srcInt;
bool flag;

extern void fn1(const Struct*) __z88dk_fastcall;

void main() {

  storage.abyte1 = 1;
  storage.abyte2 = 10;

  if (flag) {
    storage.aint = srcInt;
    fn1(&storage);

    return;
  }

  storage.aint = 1;
  fn1(&storage);
}
```

When i run this: `zcc +embedded -subtype=none -lm -compiler=sdcc -clib=sdcc_iy -m -S --list spike.c -o spike.asm  -create-app  -Cs --Werror --c-code-in-asm`

I get:
```
; ---------------------------------
_main:
;spike.c:19: storage.abyte1 = 1;
	ld	hl,_storage
	ld	(hl),0x01
;spike.c:20: storage.abyte2 = 10;
	ld	de,_storage+0
	ld	hl, +(_storage + 0x0001)
	ld	(hl),0x0a
;spike.c:23: storage.aint = srcInt;
;spike.c:22: if (flag) {
	ld	hl,_flag
	bit	0, (hl)
	jr	Z,l_main_00102
;spike.c:23: storage.aint = srcInt;
	ld	de, +(_storage + 0x0002)
	ld	hl,_srcInt
	ld	a, (hl)
	ld	(de), a
	inc	de
	inc	hl
	ld	a, (hl)
	ld	(de), a
;spike.c:24: fn1(&storage);
	ex	de, hl
	jp	_fn1
;spike.c:26: return;
	jr	l_main_00103
l_main_00102:
;spike.c:29: storage.aint = 1;
	ld	hl,0x0001
	ld	((_storage + 0x0002)), hl
;spike.c:30: fn1(&storage);
	ex	de, hl
	jp	_fn1
l_main_00103:
;spike.c:31: }
	ret
```
If you look at code generated for **;spike.c:24: fn1(&storage);**, you see that it has `ex de, hl`, to load up `hl` for the call - but `de` is not the required value of `storage` at that point.

I have been using the sdcc compiler, for no specific reason -- I kind of assumed it was more feature complete.  Is the sccz80 compiler typically the 'better' choice?  For what i can tell, the sccz80 doesn't have this issue - but given the project i am working on has gone down the path of sdcc - not sure how much effort/issues/differences I will have if I switch over - have a few inline assembly sections that would need to be changed.

Thanks
Dean.

suborb, 2021-01-21:
Thanks for the perfect test case! Ive just tried out #12016 and get the same result, so I think this is an upstream `--reserve-regs-iy` issue. If you switch to -clib=sdcc_ix then that issue will go away, though the generated code will use iy a bit too much for my liking. I've raised the upstream issue: https://sourceforge.net/p/sdcc/bugs/3169/ - we're due to pick up a new sdcc version soon (there's been a fair bit of codegen churn lately in sdcc)

sdcc vs sccz80 is personal choice. Both have their strengths and both have their place, and both will have bugs. The aim really is that code should compile and have the same end result with both compilers - they should support the same core features.

Loops are generally better with sdcc, but operations involving longs tend to be more compact with sccz80. Compilation with sdcc is usually orders of magnitude slower than sccz80 so quite a few people use sccz80 during development where you want a quick testing turnaround and sdcc for release.

suborb, 2021-01-21:
Philipp has fixed this one upstream, so until #1679 is done you could do it yourself since you're compiling from source, change:

```
index a85816291..c8fe59245 100644
--- a/Makefile
+++ b/Makefile
@@ -34,7 +34,7 @@ endif

 Z88DK_PATH     = $(shell pwd)
 SDCC_PATH      = $(Z88DK_PATH)/src/sdcc-build
-SDCC_VERSION    = 11877
+SDCC_VERSION    = 12017

 ifdef BUILD_SDCC
 ifdef BUILD_SDCC_HTTP
```

Remove bin/zsdcc and then recompile just sdcc wit:

```
make -DBUILD_SDCC bin/zsdcc
```

closed 2021-01-21 by suborb

vipoo, 2021-01-21:
That's fantastic - appreciate the quick turnaround - thank you.

feilipu, 2021-01-22:
Done the zsdcc PR #1681. Should be ok to close this once you confirm working.