z88dk/z88dk #1686: "atoi" function bug

sblendorio, 2021-01-28:
Hi, I wrote this test program:

    #include <stdio.h>
    #include <stdlib.h>

    void main(void) {
        char *str = "49152";
        unsigned int v;

        v = atoi(str);
        printf("value = %u\n", v);
    }

It should show **value = 49152**. But it depends on how it's compiled: it **works** using **clib=8080** but does *not* work under **Z80**, where it produces **value = 32767** *whatever* is the parameter.

I compiled it using these two command lines:

    $ zcc +cpm atoitest.c -o atoi-z80.com
    $ zcc +cpm -clib=8080 atoitest.c -o atoi8080.com

These are the results:

    Altair 8800 (Z80) simulator V3.8-1 build 1296 (scp created May  3 2009 at 12:19:23 with gcc 4.2.4)
    
    
    64K CP/M Version 2.2 (SIMH ALTAIR 8800, BIOS V1.27, 2 HD, 02-May-2009)
    
    A>atoi8080 
    value = 49152
    
    A>atoi-z80
    value = 32767
    
    A>

suborb, 2021-01-28:
I think you're hitting undefined behaviour!

An int (signed) holds -32768 to 32767. You're trying to parse value outside of that range. The z80 code just returns INT_MAX, whereas the 8080 version just wraps.

sblendorio, 2021-01-28:
But it's a unsigned integer... How to parse it?

suborb, 2021-01-28:
strtoul() is the more robust way.

closed 2021-01-28 by sblendorio