z88dk/z88dk #1687: double-long cast on Intel 8080

sblendorio, 2021-01-28:
Hi, I produced this little program:

    #include <stdio.h>
    #include <stdlib.h>
    #include <math.h>
    
    void main(int argc, char *argv[]) {int n = 0, i = 0;
        long xc = 6;
        long xb = 3;
        double t1;
    
        t1 = xc/(2.0*xb);
    
        printf("value=%lf\n",t1);
    }

It should print "1.0000". It works if compiled for z80 cpu:

    $ zcc +cpm -lm test.c

But, compiling it for Intel 8080, it produces a compile-time error:

    $ zcc +cpm -lm -clib=8080 test.c
    Error at file 'test.c::main' line 5: symbol 'dpush3' not defined
    Errors in source file /usr/local/share/z88dk//lib/target/cpm/classic/cpm_crt0.asm:
    Error at file 'test.c::main' line 5: symbol 'dpush3' not defined
                   ^ ---- void main(int argc, char *argv[]) {int n = 0, i = 0;

sblendorio, 2021-01-28:
tried to use these, I got this:

    $ zcc +cpm --math-dai32_8080 -clib=8080 test.c
    Error at file 'c/sccz80/l_f32_long2f.asm' line 27: symbol '___dai32_FPSIGN' not defined
    Error at file 'c/sccz80/l_f32_long2f.asm' line 31: symbol '___dai32_FPEXP' not defined
    Error at file 'c/sccz80/l_f32_long2f.asm' line 33: symbol '___dai32_BNORM' not defined
    Errors in source file /usr/local/share/z88dk//lib/target/cpm/classic/cpm_crt0.asm:
    Error at file 'c/sccz80/l_f32_long2f.asm' line 27: symbol '___dai32_FPSIGN' not defined
    Error at file 'c/sccz80/l_f32_long2f.asm' line 31: symbol '___dai32_FPEXP' not defined
    Error at file 'c/sccz80/l_f32_long2f.asm' line 33: symbol '___dai32_BNORM' not defined
    $ zcc +cpm --math-mbf32_8080 -clib=8080 test.c
    Error at file 'c/asm/___mbf32_DVBCDE.asm' line 4: symbol 'DVBCDE' not defined
    Error at file 'c/asm/___mbf32_BNORM.asm' line 4: symbol 'BNORM' not defined
    Error at file 'c/asm/___mbf32_FPADD.asm' line 4: symbol 'FPADD' not defined
    Error at file 'c/asm/___mbf32_CMPNUM.asm' line 4: symbol 'CMPNUM' not defined
    Error at file 'c/asm/___mbf32_FPINT.asm' line 4: symbol 'FPINT' not defined
    Error at file 'c/asm/___mbf32_FPMULT.asm' line 4: symbol 'FPMULT' not defined
    Error at file 'c/asm/___mbf32_SUBCDE.asm' line 4: symbol 'SUBCDE' not defined
    Errors in source file /usr/local/share/z88dk//lib/target/cpm/classic/cpm_crt0.asm:
    Error at file 'c/asm/___mbf32_DVBCDE.asm' line 4: symbol 'DVBCDE' not defined
    Error at file 'c/asm/___mbf32_BNORM.asm' line 4: symbol 'BNORM' not defined
    Error at file 'c/asm/___mbf32_FPADD.asm' line 4: symbol 'FPADD' not defined
    Error at file 'c/asm/___mbf32_CMPNUM.asm' line 4: symbol 'CMPNUM' not defined
    Error at file 'c/asm/___mbf32_FPINT.asm' line 4: symbol 'FPINT' not defined
    Error at file 'c/asm/___mbf32_FPMULT.asm' line 4: symbol 'FPMULT' not defined
    Error at file 'c/asm/___mbf32_SUBCDE.asm' line 4: symbol 'SUBCDE' not defined

suborb, 2021-01-28:
I should have fixed up the dai problem (there was some errant code leftover from when I copied mbf32 as the base) - please try again tomorrow!

For MBF32 the actual maths code isn't supplied with z88dk - you need to download an additional file and then build the library yourself.

sblendorio, 2021-01-29:
What is the download URL for the additional file?

sblendorio, 2021-01-29:
Just did a **git pull** but I got this:

    $ zcc +cpm --math-dai32_8080 -clib=8080 test.c -o test.com
    Error at file 'c/sccz80/l_f32_long2f.asm' line 27: symbol '___dai32_FPSIGN' not defined
    Error at file 'c/sccz80/l_f32_long2f.asm' line 31: symbol '___dai32_FPEXP' not defined
    Error at file 'c/sccz80/l_f32_long2f.asm' line 33: symbol '___dai32_BNORM' not defined
    Errors in source file /usr/local/share/z88dk//lib/target/cpm/classic/cpm_crt0.asm:
    Error at file 'c/sccz80/l_f32_long2f.asm' line 27: symbol '___dai32_FPSIGN' not defined
    Error at file 'c/sccz80/l_f32_long2f.asm' line 31: symbol '___dai32_FPEXP' not defined
    Error at file 'c/sccz80/l_f32_long2f.asm' line 33: symbol '___dai32_BNORM' not defined

suborb, 2021-01-29:
Sorry, okay. So:

Download URL is in this file: https://github.com/z88dk/z88dk/blob/master/libsrc/math/mbf32/z80/math_mbf32.asm

Regarding daimath you'll need to rebuild:

```
cd libsrc/math/daimath
make clean
rm ../../daimath*.lib
make
cp ../../daimath* ../../../lib/clibs/
```

sblendorio, 2021-01-29:
Just did a complete removal of z88dk and total rebuild with:

    git clone --recursive https://github.com/z88dk/z88dk.git
    cd z88dk
    sudo ./build.sh
    sudo make install

then recompiled my code with

    zcc +cpm --math-dai32_8080 -clib=8080 test.c

it worked! Thank you

sblendorio, 2021-01-29:
a question: there are reasons for preferring **mbf32** or **dai32** ?

closed 2021-01-29 by sblendorio

suborb, 2021-01-29:
On CP/M I'm not sure it makes a difference - the dai32 fp format is the same as the AM9511 FPU so in theory the library could be extended to offload certain operations - see #1543 for the benefits (or not)

If I remember correctly, the dai32math mantissa is 1 bit shorter than mbf32 but I don't think that has a serious impact.

The mbf32 implementation is certainly a lot more tested than dai32math! - It's been embedded in the ROMs of various machines and nascom basic which has been ported pretty much everywhere.

My gut says go with mbf32, but it's a personal choice - I think they perform roughly the same.

sblendorio, 2021-01-30:
is it on a different URL for license reasons?

feilipu, 2021-01-30:
> If I remember correctly, the dai32math mantissa is 1 bit shorter than mbf32 but I don't think that has a serious impact.

Actually it is a [one bit shorter 7-bit exponent](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/math/float/am9511#am9511-floating-point-format), which makes the range +63 to -64. The mantissa is identical.
This means the results are identically accurate (assuming one can rely on the calculations from both being accurate), but the DAI math format has a smaller range of +/-(2.7x10^-20 to 9.2x10^18.

> My gut says go with mbf32, but it's a personal choice - I think they perform roughly the same.

:+1: