z88dk/z88dk #1688: Stack position on CP/M

zx70, 2021-01-28:
Do I understand correctly regarding the SP initialization for the CP/M target?
Looking at the start-up code it seems to me we left it at -1.
I wonder if it should be lowered, e.g. referring to the higher byte of the BDOS vector call in the PSP (at position 0007h).

zx70, 2021-01-28:
this probably answers already: 
https://www.z88dk.org/forum/viewtopic.php?t=11004

suborb, 2021-01-28:
The -pragma-define:REGISTER_SP=NNNN has the following semantics:

* -1 leave as is
* +NNNN -> ld sp,
* -NNNN  -> ld sp,(NNNN)

So I don't think there's a need to change anything in the crt, but maybe some doc to explain how to do certain things.

zx70, 2021-01-29:
Thankyou @suborb :)
I found the answer myself too, with an quick experimental check on the Bondwell2.
The stack is already properly initialized by the CP/M system, there's no need to force it.

[image]

closed 2021-01-29 by zx70