z88dk/z88dk #1690: [CP/M] -startup=2 option fails

zx70, 2021-01-29:
The "-startup=2" option is a trick currently used on the TIKI10 to provide self-relocatable programs.
It simply sets ORG to 8000h, then a small routine at 100h moves the bytecode to the higher half of the memory and jumps there.

I suspect that the new sections (BSS, DATA..) now require something more to work correctly.
I was going to use this option as a quick win for the Bondwell BW2 graphics, then I noticed the trailing part of the TIKI's "swagfxbk.asm" file:

	SECTION		code_crt_init

	EXTERN		__BSS_END_tail
	EXTERN		__HIMEM_head
	EXTERN		__HIMEM_END_tail

	ld	hl,__BSS_END_tail
	ld	de,__HIMEM_head
	ld	bc,__HIMEM_END_tail - __HIMEM_head
	ldir

; TODO: Copy graphics to high memory

suborb, 2021-01-29:
I did some magic a while back so that the tiki100 doesn't need startup=2, just add -subtype=tiki100 and the magic will happen.

That bit of code you found relocates the code_driver section up to high memory so that the screen can be paged in.

We should probably remove -startup=2 if it's broken.

zx70, 2021-01-29:
Good, then I first need to see it working on TIKI (just to be sure everything is clean), then I'll try to sort out the rest.
I suppose it depends on the code section declaration on top of the code modules, right ?

suborb, 2021-01-29:
Yes, so the section setup is here:

https://github.com/z88dk/z88dk/blob/master/lib/crt/classic/crt_section_standard.asm#L181

And the low level primitives are here put into the appropriate section here (obviously machine specific):

https://github.com/z88dk/z88dk/tree/master/libsrc/target/tiki100/graphics

and:

https://github.com/z88dk/z88dk/blob/master/libsrc/target/tiki100/stdio/generic_console.asm

When compiled an extra output file a_HIMEM.bin is generated which appmake appends to the end of the main a.bin to create a single file on disc.

I suspect there's a z80asm directive (PHASE?) which might have made it easier (and not generated a separate file) but what's there does work.

zx70, 2021-01-29:
That explains why I couldn't just just transfer a renamed .bin file with tikidisk.

suborb, 2021-01-29:
Stick the data in the section rodata_himem or data_himem. 

I think I just had the code paging with just the primitives in high memory - gencon would have done this. 

I guess that means the +cpmdisk should probably produce a .com file as well.

zx70, 2021-01-30:
thank you!   it helped.
The startup=2 opt fails
The rndsphere.c  simplest graphics demo works, which is very good,  the available memory is now almost doubled.
dstar.c fails
now I'll try by forcing the data pages and will do a single function check..

zx70, 2021-01-31:
is there a way to assign the rodata_himem or data_himem sections in C?

suborb, 2021-01-31:
Yup:

```
#pragma codeseg code_himem
#pragma constseg rodata_himem
#pragma dataseg data_himem
#pragma bssseg bss_himem
```

zx70, 2021-02-01:
A bit of progress done, the circle and 'stencil' functions are now working.
I still miss a code_smc_himem section, I'm thinking to just add "code_smc" to the "__crt_org_graphics" special case.

suborb, 2021-02-01:
Oh, I've just realised what I did wrong!

The swapgfxbk is taking place too early - I've been assuming that the swap took place in plot so there was no need to do anything else - but the swap is taking place in the circle/draw etc so the code ends up being swapped out. Doh!

zx70, 2021-02-01:
;)
[image]

zx70, 2021-02-01:
There's a bug I couldn't solve yet on the TIKI100, regarding the point() function.
It fails when the horizontal coordinates are >255, which I do not fully understand considering that plot() works well and it is all based on the same code block: tiki100/graphics/w_pixel.asm

It can be tested using wall.c,
zcc +cpm -subtype=tiki100 -create-app  -DBANNERS -Dspritesize=16 -Dspritesizeh=8 wall.c

[image]

zx70, 2021-02-04:
[image]

DSTAR example updated

zx70, 2021-02-05:
With Bondwell Model-2  I'm stuck at this point:

```

/*
 *
 *	Bondwell Model 2 VRAM access test
 *
 *  zcc +cpm -subtype=tiki100 -create-app bw2.c
 *
 *
 *  CP/M disk image editing and starting MAME:
 *
 *  cpmrm -f bw14 bondwl02.raw  0:a.com
 *  cpmcp -f bw14 bondwl02.raw  a.com  0:a.com
 *  mame64 bw2 -flop1  bondwl02.raw
 *
 *
 */

#include <stdio.h>

main()
{

	fputc_cons(26);			// Clear screen put cursor at top-left

#asm

	JP		bw2_test
	

	SECTION		code_graphics

;   How to read the VRAM address at the current cursor position
;   -----------------------------------------------------------
;	LD		A,5            ; VRAM start addr. low
;	OUT		($21),A        ; LCD control register
;	IN		A,($20)        ; LCD data register
;
;	ld	(smc_basegfx+1),a		;  <-- a possible way to use this stuff in "pixeladdress" (to be shifted properly)
;
;	LD		A,6            ; VRAM start addr. high
;	OUT		($21),A        ; LCD control register
;	IN		A,($20)        ; LCD data register
;
;	ld	(smc_basegfx+2),a		;  <-- a possible way to use this stuff in "pixeladdress" (to be shifted properly)
;   -----------------------------------------------------------

.bw2_test

; set VRAM start addr. to point to '0'
; it is usually at the cursor position

	ld		bc,$21
	ld		a,5         ; VRAM start addr. low
	out		(c),a
	xor		a           ; set to 0
	dec		c
	out		(c),a
	
	inc		c
	ld		a,6         ; VRAM start addr. high
	out		(c),a
	xor		a           ; set to 0
	dec		c
	out		(c),a
	

; save current bank 

	in      a,(2)     ; PPIC, get current bank
	ld      (bw2_currentpage+1),a

	ld      a,1		; no mask, to keep the NMI disabled
	out     (2),a     ; PPIC, set VRAM bank
	
	ld      a,1		; no mask, to keep the NMI disabled
	out     (2),a     ; PPIC, set VRAM bank
	
	// Fill VRAM with 4 different pattern stripes
	ld      hl,0
	ld      de,1
	ld      (hl),$5F
	ld      bc,4000
	ldir
	ld      (hl),$F0
	ld      bc,4000
	ldir
	ld      (hl),$FA
	ld      bc,4000
	ldir
	ld      (hl),$F0
	ld      bc,4000
	ldir

.bw2_currentpage
	ld		a,0
	out		(2),a          ; PPIC, restore previous memory bank
	
	; Program end.
	; RET is probably optional, the tail portion will be relocated ;)
	ret

	SECTION		code_crt_init

	EXTERN		__BSS_END_tail
	EXTERN		__HIMEM_head
	EXTERN		__HIMEM_END_tail

	ld	hl,__BSS_END_tail
	ld	de,__HIMEM_head
	ld	bc,__HIMEM_END_tail - __HIMEM_head
	ldir

#endasm

}

```

I'm not able to stabilize it enough to make putpixel() work.
e.e. the second picture shows the picture shifted due to the different shift of the "scroll register" when the cursor is at a different position.

[image]

zx70, 2021-02-06:
one more alert, the normal bw2 bios alters the IX register when printing a character, and alters few bytes around INT and NMI, e.g. to deal with the disk drive events

suborb, 2021-02-06:
That altering of ix is very antisocial! 

I'm guessing we should push/pop in fgetc_cons() so that we don't break sdcc.

zx70, 2021-02-06:
Same problem with fputc_cons(), bdos(), bios() and possibly hardcodec jumps somewhere else in the lib. 
That tempted me at trapping the bdos() entry chamging the ptr at address 6 to step by a custom  wrapper we might provide. 
BTW it could be even worse.   It does not solve for the BIOS calls, and moves the official BDOS position limiting the tricks to probe the CP/M memory map.  On the positive aspects,  it can be restricted to  the few targets needing it (the Bondwell 2 could be the only case) .

suborb, 2021-02-06:
A little grep finds:

- fcntl uses bdos()
- stdio does call 5
- time calls 5, but saves and uses ix itself

Given that it is CP/M I think we have to solve it generically for our libraries (so not just a Bondwell2 option) - I'm wondering if a BDOS interceptor is needed - replace all call 5/jp 5 with `call __z88dk_bdos_interceptor`? - I don't like the idea of changing the bdos vector down in page 0.

zx70, 2021-02-07:
It's a bit an extra rule for the developers to follow,  but looks clean. 
I saw JP 5 elsewhere but in target specific code (e.g. the crap graphics code for the Kaypro4)

zx70, 2021-02-10:
Stability problem found !
Bondwell 2 target will follow soon :)

zx70, 2021-02-10:
[image]
putsprite() still needs to be fixed, but most of the library is now ready off-the-shelf.
@suborb, after the obvious hassle, I'm really loving the way you sorted out the graphics libraries !

zx70, 2021-02-10:
[image]
Switching to w_putsprite3.asm did the trick.

zx70, 2021-02-11:
Target specific version or w_putsprite() is now ready for Bondwell 2.
No console extras at the moment.
The Tiki target should be checked more accurately (e.g. clg() could enter in hirez mode to transparently enable the monochrome stuff.).
BTW the average lib status is now clean ..and cool  :)

I'm removing "-startup=2" because not compatible with the new relocation methods.

closed 2021-02-11 by zx70