z88dk/z88dk #1698: (zcc) Processing of m4 files changed so that results are not written back to user directory

aralbrec, 2021-02-10:
This is referencing the [Sep 26, 2020](https://github.com/z88dk/z88dk/commit/302a35594a7eddc4715605428cf4ade201ddfc52#diff-4b7f65126a1a019009a7156d5dea1e665612f7bb3172ee987a31156b5e88ed95) commit into zcc.

There appears to be a change where processed m4 files are not written back to the user directory.  So `foo.asm.m4`, `foo.inc.m4`, `foo.c.m4`, `foo.h.m4`, etc will not see their result files written back.

It was deliberately done this way to allow includes from source files to continue to work.  For example, if a header was constructed by m4 so that the file begins life as `foo.h.m4` then the generation of `foo.h` into the user's directory would allow c code to include the file with `#include "foo.h"`.  If the result is not written into the user directory, such an include wouldn't work and you'd be forced to invoke m4 directly to create such files beforehand rather than allowing zcc to automatically deal with them.

suborb, 2021-02-10:
These changes were done to permit concurrent builds, though this issue was in inadvertent side effect: I haven't seen any projects that work in this way.  

I guess we have the following scenarios (thinking about single line compilations since tracking dependencies in a makefile handles this issue):

* m4 processing a file with the z88dk tree
* m4 processing a project file

I'm struggling to see a sensible scenario where a `.c.m4` file is supplied to `zcc`, compiled and then another project file includes that `.c` file - can you give an example of it?

The preprocessing of files within the z88dk tree is already covered (eg crt files) - these are .asm files so end up in zccopt tempdir and that directory is added to the assembler search path. This has the happy effect of not mutating the tree.

For files outside, they also end up in zccopt tempdir but that directory isn't added to the preprocessor search list,  so I think `.asm` and `.inc` files should work correctly (unless they need to be renamed back to the original basename)

Just `.h` files left then, the simple solution would be to add the temporary directory to the preprocessor include path (along with any potential rename)

Of course if you have multiple files of the same name that need preprocessing then you may well have bigger problems!

aralbrec, 2021-02-10:
People do the strangest things :)

https://github.com/mojontwins/MK1/blob/master/src/dev/mainloop.h#L98

C source is embedded in header files which are included into other .h and .c files.  Even though it's really bad practice, c files will get included into other c files.

Including asm files into other asm files is very common among tools without linkers and many people who are used to such tools will do the same in z88dk even though there's a much better way to do things.  A master `main.asm` includes the entire project composed of many files and then the assembler builds the output from one input file.

Multiple files of the same name is not unusual I don't think so I'm not so sure mutating the tree is avoidable.  Concurrent builds are important too but perhaps that needs to be under user control rather than limiting what the tools will do?  If you want concurrent builds, create the result of the m4 files beforehand?

suborb, 2021-02-10:
I think key thing here is that C source can be included wherever you like - nothing has changed there and it behaves the same as with any compiler.

What we're talking about here is the extra m4 translation step that gets silently applied - that really ought to be a separate manually driven process: yes I've been caught out and edited the wrong file and wondered why my changes didn't get applied.

I can restore the behaviour, but it's certainly esoteric and unique to z88dk and the behaviour will result in intermittent, un-reproducible errors when people try to utilise their multi-core CPU with a `make -jX`, so I'd propose to be able to opt into this behaviour (copying back processed files silently) rather than automatically doing it.

BTW, By tree I meant z88dk rather than the user project

suborb, 2021-02-10:
I've added `-copy-back-after-m4` which restores the functionality but you have to "opt into it".

closed 2021-02-10 by suborb