z88dk/z88dk #1701: build problems

SM7I, 2021-02-15:
Having issues buildning z88dk. Have latest nightly and uthash, but gets this all the time:

gcc -MMD -I. -Ilib -It -g -Wall -std=gnu11 -O3 -I../common -I../../ext/optparse                   -I../../ext/regex -I../../ext/uthash/src  -g -O2 -c -o ../common/objfile.o ../co                  mmon/objfile.c
../common/objfile.c: In function ‘file_rename_sections’:
../common/objfile.c:1041:8: error: ‘REG_OKAY’ undeclared (first use in this func                  tion)
     == REG_OKAY)
        ^~~~~~~~
../common/objfile.c:1041:8: note: each undeclared identifier is reported only on                  ce for each function it appears in
../common/objfile.c: In function ‘file_add_symbol_prefix’:
../common/objfile.c:1131:76: error: ‘REG_OKAY’ undeclared (first use in this fun                  ction)
  ((reti = regexec(&regex, utstr_body(symbol->name), 0, NULL, 0)) == REG_OKAY) {                   // match
                                                                     ^~~~~~~~
../common/objfile.c: In function ‘file_change_symbols_scope’:
../common/objfile.c:1240:76: error: ‘REG_OKAY’ undeclared (first use in this fun                  ction)
  ((reti = regexec(&regex, utstr_body(symbol->name), 0, NULL, 0)) == REG_OKAY) {                   // match
                                                                     ^~~~~~~~
Makefile:71: recipe for target '../common/objfile.o' failed
make[1]: *** [../common/objfile.o] Error 1
make[1]: Leaving directory '/root/z88dk_git/z88dk/src/z80asm'
Makefile:140: recipe for target 'bin/z88dk-z80asm' failed
make: *** [bin/z88dk-z80asm] Error 2

suborb, 2021-02-15:
I think you're missing the submodules? - https://github.com/z88dk/z88dk#installation

SM7I, 2021-02-15:
Shouldn’t they be included with a git clone ?

Hämta Outlook för iOS<https://aka.ms/o0ukef>

suborb, 2021-02-15:
Unfortunately they aren't.

To clone with submodules use git clone --recursive https://github.com/z88dk/z88dk.git. To add the submodules to an already existing clone use git submodule update --init --recursive

SM7I, 2021-02-15:
It seems to have solved the problem. Thanks a bunch !!

Med vänliga hälsningar: Johan Engdahl
-----------------------------------------------------
Email: sm7i.ham@gmail.com
BBS: sm7i@sk7es.m.swe.eu
Tel: 070-3277808

closed 2021-02-15 by suborb