z88dk/z88dk #1702: zx0 - newlib target

feilipu, 2021-02-16:
I'm trying to convert the rc2014 and yaz180 targets to use zx0 where relevant.
But, it is not working. The system hangs, doesn't boot. 

I am suspicious that the zx0 compression routine is not being properly called because the compression is silent during generation of the ROM.

With zx7 I get this:
```sh
phillip@SilverQuad:~/Z80/RC2014/ROMs/CPM-IDE/acia$ zcc +rc2014 -subtype=acia -SO3 -llib/rc2014/ff_ro --math32 --list -m --max-allocs-per-node2000000 @cpm22.lst -o ../rc2014-acia-cpm22 -create-app
ZX7: Optimal LZ77/LZSS compression by Einar Saukas
File converted from 851 to 241 bytes! (delta 10)
phillip@SilverQuad:~/Z80/RC2014/ROMs/CPM-IDE/acia$ 
```

I would expect something similar having converted to `__crt_model` 3, but the build is silent.
```sh
phillip@SilverQuad:~/Z80/RC2014/ROMs/CPM-IDE/acia$ zcc +rc2014 -subtype=acia -SO3 -llib/rc2014/ff_ro --math32 --list -m --max-allocs-per-node2000000 @cpm22.lst -o ../rc2014-acia-cpm22 -create-app
phillip@SilverQuad:~/Z80/RC2014/ROMs/CPM-IDE/acia$ 
```

In the CRT listing the correct call to `asm_dzx0_standard` is found.
```asm
20    0070                 IF (__crt_model >= 2)
21    0070              
22    0070                    ; rom model + data section is compressed
23    0070              
24    0070                    EXTERN __CODE_END_tail
25    0070              		EXTERN __DATA_head, __DATA_END_tail
26    0070              
27    0070  21 00 00          ld hl,__CODE_END_tail
28    0073  11 00 00          ld de,__DATA_head
29    0076  01 00 00          ld bc,__DATA_END_tail - __DATA_head
30    0079              
31    0079  78                ld a,b
32    007A  B1                or c
33    007B              
34    007B                  IF (__crt_model = 2)
35    007B                    EXTERN   asm_dzx7_standard
36    007B                    call nz, asm_dzx7_standard
37    007B                  ELIF (__crt_model = 3)
38    007B                    EXTERN   asm_dzx0_standard
39    007B  C4 00 00          call nz, asm_dzx0_standard
40    007E                  ENDIF
```

suborb, 2021-02-16:
Have you recompiled appmake? In an old environment I had your problem, but as soon as new appmake was dropped in I could see the compression working:

```
zcc +rc2014 -subtype=acia world.c -pragma-define:CRT_MODEL=3  -m -create-app
world.c:11: warning 59: function 'main' must return value
ZX0 v1.5: Optimal data compressor by Einar Saukas
[...............................................]
File compressed from 482 to 98 bytes! (delta 2)
```

feilipu, 2021-02-16:
Yes, rebuilt several times. But I suspect I may need to delete the binary to force a full rebuild. And will comment then.

Could it be possible to add a clean binary option to the makefile, if it doesn't already exist?

feilipu, 2021-02-16:
phillip@SilverQuad:~/Z80/RC2014/ROMs/CPM-IDE/acia$ zcc +rc2014 -subtype=acia -SO3 -llib/rc2014/ff_ro --math32 --max-allocs-per-node2000000 @cpm22.lst -m -o ../rc2014-acia-cpm22 -create-app
ZX0 v1.5: Optimal data compressor by Einar Saukas
[...............................................]
File compressed from 856 to 220 bytes! (delta 3)
phillip@SilverQuad:~/Z80/RC2014/ROMs/CPM-IDE/acia$

closed 2021-02-16 by feilipu