z88dk/z88dk #1704: memset() generates "incorrect" code when run on zxn and accessing layer 2 screen as bank 0

ncot-tech, 2021-02-18:
Trying to use this routine to clear the Next's layer 2 screen results in garbage being displayed on the screen. 
The Next's screen is banked into bank 0 which occupies the bottom 16k of address space and is write only, reading from there reads the ROM.

```
IO_123B = 0x03;
memset(0, 0, 16384);
```

The memset() gets turned into this Z80

```
ld    a,0x03
ld    bc,_IO_123B
out    (c),a
ld    hl,0x0000
ld    (hl),0x00
ld    e, l
ld    d, h
inc    de
ld    bc,0x3fff
ldir
```

And the reading that LDIR does reads from the ROM and not the screen RAM it thinks it's accessing.

suborb, 2021-02-18:
That's a fun one! I had something similar on the Camputers Lynx a while back.

`memset()` is being replaced by a compiler builtin, hence it's being placed inline. I don't know if there's a way to persuade sdcc that the the memory being pointed to is write only and replace the ldir code with a different construct. 

The only alternative would be to lose the inlining and call a non-optimal memset style function. Unfortunately I can't find one of those in the library either.

Adding one wouldn't be a problem but I am curious to know if there is a compiler only solution.

suborb, 2021-02-21:
There's no standard way to indicate writeonly memory and it looks like there's not an sdcc extension, so I've added `memset_wr()` which will skip the reading bit.

It's a function, so isn't inlined but hopefully it'll suit your purposes.

closed 2021-02-21 by suborb

ped7g, 2021-02-22:
@suborb 
> There's no standard way to indicate writeonly memory
not an expert on C, so I may have dumb question, but what about `volatile`, if the pointer would be tainted by it? It's not "write-only" either, but it's then "can read anything else so should not read", so the `ldir` memset shouldn't be used for that either.  But I have no idea if `volatile` taint is handled by z88dk compilers.

suborb, 2021-02-22:
That's a good point, and it's one of the things I tried since I thought it should work, this for example:

```
#include <string.h>

void func() {
        volatile char *ptr = 0;
        memset(ptr, 0, 16384);
}
```

Compiles without grumbling with sdcc, with clang you get a non-fatal warning about dropping the volatile so even there it's not foolproof.