z88dk/z88dk #1706: Fastcall on gbz80

basxto, 2021-03-01:
I played a bit around and wanted to see, what fastcall generates.

C code used:
```c
#define uint8_t unsigned char

uint8_t stupid(uint8_t x) __z88dk_fastcall{
  return (uint8_t)((uint8_t)8 + (uint8_t)x);
}

void main(){
    stupid(5);
}
```

On z88dk (sccz80) this function becomes:
```asm
._stupid
	push	hl		; 1  16
	ld	h,0		; 2  8
	ld	bc,8		; 3  12
	add	hl,bc		; 1  8
	ld	h,0		; 2  8
	add	sp,2		; 2  16
	ret			; 1  16
				;=12 84
```
and is called as
```asm
	ld	hl,5		; 3  12
	call	_stupid	; 3  24
				;=6  36
			;TOTAL	 =18 120
```
On vanilla sdcc without fastcall this function becomes:
```asm
_stupid::
	ldhl	sp,	#2	; 2  12
	ld	a, (hl)	; 1  8
	add	a, #0x08	; 2  8
	ld	e, a		; 1  4
	ret			; 1  16
				;=7  48
```
and is called as
```asm
	ld	a, #0x05	; 2  8
	push	af		; 1  16
	inc	sp		; 1  8
	call	_stupid	; 3  24
	inc	sp		; 1  8
				;=8  64
			;TOTAL	 =15 116
```

For a single call vanilla sdcc is the faster and smaller, but I think z88dk could be a lot faster and smaller.

-----
Does z88dk always have to do integer promotion?

Can't it be assumed for 8bit parameters that the upper byte is always 0?
I would either assume it being 0 or set it 0 in the function but just do `ld l,5` for the call since the value of h doesn't matter anyways.

Why is `hl` pushed? I don't see why that should be done

`pop` would be smaller and faster than `add	sp,`

-----
Effectively it could do something like
```asm
._stupid
	ld	h,0		; 2  8
	ld	a,l		; 1  4
	add	a,8		; 2  8
	ld	l,a		; 1  4
	ret			; 1  16
				;=7  40
```
and
```asm
	ld	l,5		; 2  8
	call	_stupid	; 3  24
				;=5  32
			;TOTAL	 =12 72
```

Even sticking to integer promotion can be smaller and faster:
```asm
._stupid
	ld	h,0		; 2  8
	ld	bc,8		; 3  12
	add	hl,bc		; 1  8
	ld	h,0		; 2  8
	ret			; 1  16
				;=9  52
			;TOTAL	 =14 84
```

suborb, 2021-03-01:
Thanks for your comments, yup there's a couple of things that can be done there.

The `push` can be explained by sccz80 being single pass, it doesn't have a register allocator or analyser, so it has to preserve the fastcall variable at the start of the function since at that point in time it doesn't know how many times it's going to be used within the function.

The corresponding `add sp,2` can be easily swapped out for a pop - there's just a range limit that's incorrect.

The promotion probably doesn't have to be done if the destination is also an 8 bit datatype, and I do think there's probably too much promotion going on there.  Internally, the working unit is mostly a 16 bit value: there are some places (eg around z180 mult) that we skip the promotion - it looks like that logic should be extended to other operations.

basxto, 2021-03-01:
Avoiding some 16bit promotion might reduce the pressure on HL.

This register pair has to be used to access variables and parameters who got placed on the stack, since there is no IX and IY like on z80.

Using HL for the first parameter and return value probably worsens that situation.

spth, 2021-09-19:
Vanilla SDCC has changed the default calling convention for gbz80; it now compiles this to:

~~~~
_stupid::
	add	a, #0x08
	ret

_main::
	ld	a, #0x05
	jp	_stupid
~~~~

suborb, 2021-09-19:
So by default the sdcc and gbz80 abi has changed?

How does this change affect the traditional calling convention? 

We deliberately didn’t use fastcall on gbz80 since it wasn’t supported by sdcc.

spth, 2021-09-19:
SDCC has implemented a new calling convention for gbz80 an while ago, and the new one is now the default (since about two weeks ago).
This can be changed using the command line option --sdcccall 0 to make the old one the default again. Individual functions can be marked for the old or new calling convention (overriding the default) using __sdcccall(0) and __sdcccall(1).
In source code, the predefined macro __SDCCCALL can be used to check for the default (when the macro is undefined or defined to 0, the old is the default, when it is defined to 1, the new is the default).

feilipu, 2022-05-16:
Lots of things improvd, rewriting gbz80 libraries, and sdcc side improvements too, so closing this pending a specific enhancement suggestion.

closed 2022-05-16 by feilipu