z88dk/z88dk #1708: (classic) MSX2 support

suborb, 2021-03-02:
We've got pretty extensive support for the MSX1, so it's probably time to take a look at the MSX2.

The main difference seems to be more graphics modes and some interesting hardware accelerated drawing operations.

Let's start off with:

- [ ] Offloading graphics operations
- [ ] Gencon

And come back to sprites and other APIs when I know more about the machine.

artrag, 2022-02-01:
Be aware also of the fact that the new VDP has less restrictive timing.
On msx2 (with a plain z80 @3.5MHz) you do not need to wait between consecutive VRAM accesses.
There is an overall I/O bandwidth increase for this reason.

suborb, 2023-03-25:
Having fun with this. I've added SCREEN 0: TEXT 80, SCREEN 4 and SCREEN 5.

SCREEN 5 is where I've been having fun. Low level plot/point primitives are done with VDP commands and the speed is meh - nothing too special. Using the VDP for lines is much more impressive. In the video the board in mode 5 is drawn with lines, and the pieces drawn using plot.

Text terminal scrolling in SCREEN 5 is done by a YMMM command so is speedy. Printing characters is basically a copy of the SAM code, but writing to the VRAM byte by byte. I think HMMC could be used but it might slow things down since we'd write to a RAM buffer and then blit. 

SCREEN 6 and the 512 wide modes throw a spanner in the works: the VDP graphics uses the narrow flavour, rather than switch to wide I think I'll force fatpix since that way the aspect ratio is sensible.

https://user-images.githubusercontent.com/394721/227745556-79d05a3a-99c4-49b9-ae65-387afc4f75f6.mov

And the almost obligatory mode 5 Mandelbrot:

[image]

suborb, 2023-03-26:
I've left out screen 7 for the moment - easy to add if needed.

I'm going to (finally!) close out this issue since we've ostensibly got support now. I'm sure there are improvements and extensions that could be added, but I'm going to look at +cpc graphics next.

closed 2023-03-26 by suborb