z88dk/z88dk #1709: [SAM Coupé] Support for Sam C (small C) programs

dandoore, 2021-03-02:
I have a lot of stuff I enjoyed doing in [Sam C](https://www.worldofsam.org/products/sam-c) back in the day which I'd like to pick up again in a modern cross-compiled way but there is currenly limited only support for driving the hardware e.g. nailed to MODE1, characters only, no support for extra Sam functions etc. 

Both assembler and compilers are way beyond me but I did extract the libraries and headers [SamSmallC.zip](https://github.com/z88dk/z88dk/files/6069142/SamSmallC.zip) from the Sam C disk  
[SAM-C.zip](https://github.com/z88dk/z88dk/files/6069155/SAM-C.zip) which might help as it contains the assembler code for various functions and calls?

suborb, 2021-03-02:
That's odd. I was thinking about adding MGT disc support last week (but I didn't do it)

Yes, SAM support is a little lacking: I think a lot of stalled because the ROM didn't provide support for, for example a POINT, function so we'd need to implement the graphics primitives ourselves.

This stalled as a result of the banking possibly paging out the graphics code - we've now figured out how to handle that on other machines so there's no technical obstacles to improving support.

dandoore, 2021-03-02:
That's awesome, I have had an intial go at hacking some things to work under zcc by dragging in bits of the Sam-specific includes ASM into the files just to get it working, the differnces in stdio are a stumbling block at the moment (the Sam one looks a bit bespoke) but that might just be my unfamiliarity with z88dk showing.

I was pretty pleased getting this up though :-)
[image]

dandoore, 2021-03-02:
> Yes, SAM support is a little lacking: I think a lot of stalled because the ROM didn't provide support for, for example a POINT, function so we'd need to implement the graphics primitives ourselves.

If you mean a POINT(x.y) for returning a pixel colour value then Sam C used 'getpixel' (I remember using it back in 1996..)

int getpixel(int x, int y)
{
    #asm
    pop bc
    pop de
    pop hl
    push hl
    push de
    push bc
    ld  a,191
    sub e
    ld  d,a
    ld  e,l
    ex  de,hl
    scf
    rr h
    rr l
    push af
    call 0x3fbb
    ld l,a
    pop af
    ld a,l
    jr c,bezpod
    rrca
    rrca
    rrca
    rrca
bezpod: and 15
    ld l,a
    ld h,0
#endasm
}

This is one of the things in misc.c in that zip file,

dandoore, 2021-03-13:
I've started a (very rough) update of the tools to make MGT (formerly DSK) files and launch them in SimCoupé called [mgtman](https://github.com/dandoore/mgtman) and made a simple compile-makedisk-lauch batch script for Windows.

suborb, 2021-03-15:
I've not forgotten about this, it's next on my radar. I'm still undecided whether to use the ROM or take over the entire machine. The problem with the ROM is that STREAMS is located at 0xb000 which limits the size of the program.

It might be better to take over the entire machine, and use a bootstrap to load the program into the appropriate page. Any ROM routines could be called via trampoline (eg for file access). I need to find some example code for the paging since it confuses me at the moment.

* [ ] Gencon driver (so ansiterminal comes in via that)
* [ ] Screen mode switching
* [ ] User font
* [ ] 36 UDGs (we'll reserve code 165 for underline/bold tricks)/128 if not ROM
* [ ] vpeek()
* [ ] Plot/reset/xor/point pixel
* [ ] Appmake MGT support

stefandrissen, 2021-03-15:
What do you need for paging? Paging is a bit unwieldy on the SAM Coupé. There are four 16k pages (ABCD), but you can only set A (and B is the next page) or C (and D is the next page).

I have a some constants over at https://github.com/stefandrissen/SAM-MOD-player/blob/master/src/ports.i which may help.

suborb, 2021-03-15:
Yes, it looks fairly unpleasant to deal with. And thank you for confirming what I was reading but not quite believing about the paging.

So, at the moment the SAM target loads to 0x8000, if we use the ROM then it looks like we have a hardstop at 0xb000 otherwise we trample the STREAMs variables. We could locate lower, but it's bit awful to only support max (non banked) program size of ~20k on a 256k machine.

The alternative is probably to use a bootstrap and then load our code into banks, with ROM calls via an interposer that restores the paging to the BASIC environment.

If we want to manipulate the screen directly, then we have to page in the screen to C+D. Which tramples our code. Which makes me think that going for an allram setup might be better.

I'm still undecided what's the best approach, so I'm sort of canvassing for opinion.

stefandrissen, 2021-03-15:
Everything I have ever coded sits in A+B (no ROM) with screen in C+D since it makes interrupts nice and easy to use (IM1) and also allows normal trapping of NMI.

suborb, 2021-03-15:
Thank you, I think that's what I wanted to hear!

frodete, 2021-03-15:
Good to see that someone is taking on the mantle for the SAM port.  When I started out, Z88dk lacked support for paging infrastructure, the graphics primitives were being worked on and other stuff were also in the works which led me to just wait it out - forgetting about it all together.

I believe I had a rough sketch with section A+B for code, C+D for easy access to screen and the top 8K for misc routines (paging, trampoline, fast screen routines, stack, etc.) so that ROM0 could be paged into section A to access stuff like the jump table (just to not have to code them all again) and DOS (which goes into section B).

I'm pretty sure the ROM does not use 0xb000 for anything special, particularly STREAMS - unless I'm forgetting something crucial here.  A screen is normally paged in at 0x8000. ROM1 goes into 0xc000, but this is normally handled transparently by ROM0.

Right now I'm knee deep in work, but in a few months, I hope I have some spare time.

stefandrissen, 2021-03-15:
> Thank you, I think that's what I wanted to hear! Do you mind if I crib a bit of bootstrapping from your code?

Please do.

stefandrissen, 2021-03-15:
> I'm pretty sure the ROM does not use 0xb000 for anything special, particularly STREAMS - unless I'm forgetting something crucial here. A screen is normally paged in at 0x8000. ROM1 goes into 0xc000, but this is normally handled transparently by ROM0.

I had never seen this streams thing before either, but when looking in the [technical manual](https://sam.speccy.cz/systech/sam-coupe_tech-man_v3-0.pdf) found that it is documented in the memory map on page 27 (pdf page 32). In the [SAM ROM disassembly](https://github.com/simonowen/samrom/search?q=CHANS) I only see CHANS being assigned to the more normal 5CB6... at which point the unfortunate layout of the memory map in the technical manual is revealed.

The start of system variables is 0x5a00 (can be confirmed by ``print hex$(svar0)``)
The start of channels is 0x5cb6 and can supposedly go up to 0xb000, which is complete madness. I am guessing that the 0B000H is a typo and that it should actually read 08000H.

frodete, 2021-03-16:
> I had never seen this streams thing before either, but when looking in the [technical manual](https://sam.speccy.cz/systech/sam-coupe_tech-man_v3-0.pdf) found that it is documented in the memory map [...]

You are correct.  The PDF does have a typo and the text is also misaligned. 05CB6H should be START OF CHANNELS.

stefandrissen, 2021-03-16:
> You are correct. The PDF does have a typo and the text is also misaligned. 05CB6H should be START OF CHANNELS.

Yes, that's what I meant with "unfortunate layout", "unfortunate typesetting" may have been less ambiguous in this case.

suborb, 2021-03-16:
Honestly, I've not looked at a manual before adding a target in ages, and now I remember why!

Thank you all, things are now falling into place and good progress is being made. Though I'm sure I'll miss some obvious optimisation tricks.

I've taken the easy way out at the moment and I've got the screen in the lower 32k whilst I add features. Hopefully by the weekend it should have a fairly good level of support beyond the existing +zx compatibility.

suborb, 2021-03-17:
Current state of play:

- [x] Appmake now creates mgt discs (this has been merged already)
- [x] gencon for all screen modes (colour, vpeek, scrolling)
- [x] plot/unplot/point/draw/circle etc for all screen modes
- [x] One bit sound should be there from before
- [ ] ROM based ANSI terminal to be removed and gencon bridge enabled
- [ ] Keyboard reading (will repurpose the inkey +zx and add the extra keys)
- [ ] stencil/putsprite drawing routines 
- [ ] Joystick
- [ ] BASIC crt0 to be restricted to ROM routines only (fudge a linking error when using direct hardware access)
- [ ] Disc/file access
- [ ] Low memory crt0, handling nmi
- [ ] SAA1099 support (is there a frequently used tracker engine that can be imported?)
- [ ] arch/sam.h defining SFRs for direct hardware access

stefandrissen, 2021-03-17:
>     * [ ]  Joystick

Joysticks are ZX Spectrum 'Interface 2' style - mapped to keys 67890 and 12345

>     * [ ]  SAA1099 support (is there a frequently used tracker engine that can be imported?)

https://github.com/stefandrissen/e-tracker-player

suborb, 2021-03-19:
My status comment seems to have gone awol.

Anyway, I've merged what I've done so far, the current feature set is here: https://github.com/z88dk/z88dk/wiki/Platform--Sam-Coupe.md - There will be some rough spots, probably some non-working bits and definitely some sub-optimal bits so feedback/patches will be gratefully received.

Although I've merged this initial work, I'm planning to carry on adding more features.

dandoore, 2021-03-20:
Very, very nice work :)

The section under quickstart might need a tweak: The MGT file will need to have 'samdos2' code in directory entry 1 and then the code produced by the compiler as 'autoi.bin' as the bootstrap process looks for 'auto*' to load after the DOS.  Also if the auto.bin is set to execute then it will just all load and run without having to issue the CALL command.

suborb, 2021-03-20:
I’ve obviously not made it clear so it needs a reword. 

The file auto.bin is the Samdos file. I’ll rename the option to something else to make it more obvious as well. 

Autoexec is on my radar - I’m just making the code reusable for +D discs as well.

dandoore, 2021-03-20:
Samdos needs to just be Samdos though it could actually be code for MasterDOS, BDOS, MasterBasic+DOS depending on what you want to run) - the auto.bin is the code dropping on so DOS cannot be named 'auto*'.

dandoore, 2021-03-20:
> Autoexec is on my radar - I’m just making the code reusable for +D discs as well.

If it helps,  I just added Autoexec stuff into [MGTman](https://github.com/dandoore/mgtman)'s SaveFile routine.

frodete, 2021-03-20:
This is stellar work!  Congratulation on the effort.  Just spottet my crt0 code - from 2002.  I feel old.

suborb, 2021-03-20:
> Samdos needs to just be Samdos (though it could actually be code for MasterDOS, BDOS, MasterBasic+DOS depending on what DOS you want to boot the machine into) - the auto.bin is the code dropping on so DOS cannot be named 'auto*'.

Okay, I ended up confused there! I ended up renaming SAMDOS2 to auto.bin and making it executable because:

a) I misread the batch file
b) It made the no auto file warning go away
c) It worked

Hopefully it's now correct and auto-booting application discs are created by default.

If the SAMDOS2 file can be distributed without restrictions I'm happy to include it by default and simplify things even further.

> This is stellar work! Congratulation on the effort. Just spottet my crt0 code - from 2002. I feel old.

Thanks! There's still more to do though. Did you ever write those ROM calling trampolines you were talking about above? - I'd like to borrow them if so.

dandoore, 2021-03-20:
> If the SAMDOS2 file can be distributed without restrictions I'm happy to include it by default and simplify things even further.

Yes, it's freely distributable - @stefandrissen has the source over here if you're so inclined:) https://github.com/stefandrissen/samdos

suborb, 2021-03-20:
> Yes, it's freely distributable - @stefandrissen has the source over here if you're so inclined:) https://github.com/stefandrissen/samdos

Done: I've just included the blob rather than assembling from source. Creating an auto-booting disc is now a simple case of:

     zcc +sam file.c -create-app

Which is great and matches the options needed for other targets.

frodete, 2021-03-20:
> Thanks! There's still more to do though. Did you ever write those ROM calling trampolines you were talking about above? - I'd like to borrow them if so.

Sorry, no.  They were just initial thoughts - 20 years ago...

suborb, 2021-03-21:
For the moment, I think I'm done - so I'm going to close off this issue. Please raise any enhancements/bugs via new issues.

Over the weekend, etracker has been imported and the default subtype switched (again!) so that 64k of address space is available for user programs.

The biggest omission in terms of core functionality is file IO but I don't feel like writing all the random access routines required at the moment!

closed 2021-03-21 by suborb