z88dk/z88dk #1711: PC-8801 integration

bferguson3, 2021-03-06:
Hello,

I am working on this:
https://github.com/bferguson3/pc88-c

It uses SDCC and a simple Makefile, Python scripts and a few dependencies (no native binaries) to create a 2D disk image that auto-loads and boots in an emulator. I've also successfully written the image to a physical disk using Xdisk2 and booted it on my personal PC-8801mkII FR. (The sprite example in V2 mode is rock-solid!)

I also translated a bunch of information on the hardware and video processor. 
Now that I made a clean-but-simple Make, it should be easy to adapt to add support for / include in z88dk. I am somewhat unfamiliar with z88dk though, so where to start and any suggestions or questions are welcome.

Thanks :)

suborb, 2021-03-06:
It's been a while since we did some real work on the PC88 (#996 )

The state of the pc88 world in z88dk is here: https://github.com/z88dk/z88dk/wiki/Platform---NEC-PC8801, the target specific code is here: https://github.com/z88dk/z88dk/tree/master/libsrc/target/pc88

Essentially we've got:

* Text mode text printing (without any colour support)
* Semi-graphic plotting, lines, circles etc
* Hires text printing including UDGs - this time with colours
* Hires plotting, lines, circles, 1 bit sprites
* Interrupts
* PSG support
* 1 bit beeper support

And all the usual stuff you get for "free" with z88dk (compression, data structures, maths etc). 

This can be compiled together to create either a tape image (so loaded from basic) or into an IPL d88 disc image using whichever compiler you want. It looks like we're not using an allram mode for IPL for some reason. 

If I remember correctly, the PC88 has got some interesting video hardware tricks which I know we aren't using (we brute-forced the hires graphics) - so looking at your docs would be useful to improve support there.

I'm not too clear what you're suggesting, but it does seem like the best approach is to review what you've done and see what bits can be added to the z88dk libs - we've got a fair bit of support but we're not experts on the machine so we will have missed a few tricks and the target probably needs a bit of cleanup.

bferguson3, 2021-03-06:
Hey suborb, thanks for the reply!

I tried the PC88 branch of z88dk and honestly was pretty underwhelmed,
perhaps because I couldn't figure out how to make a working disk, so I
decided to just start from scratch.

It sounds like I might have a bit more progress on the underlying bits
(like graphic modes) and disk access, maybe, without the zdk benefits.

I couldn't figure out how to make an IPL d88 image with z88dk, only a way
to make a tap image to be loaded with mon. (This is a pretty archaic method
of loading I think, only really used on 6001, and can be made more user
friendly with a Basic stub and RUN"CAS:, I think?)

My toolset is all Python for portability, including a tool that has the
same basic functionality as the D88SAVER.EXE tool for making disks and
adding IPL and files.

I couldn't parse the draw stuff so I don't know how z88dk handles the
screen. I have the VDP/ALU and graphics drawing part down pretty well for
all modes, so I can help here.

What I need to do next is implement V2 mode draw calls (right now the
bitmap call uses V1), research N88 Basic so BIOS calls can be used
effectively, finish disk save functionality (start is in src/disker.z80),
and add sound. PSG and beeper is great, but OPN/A would let me feel support
is complete.

Sounds like all of that to-do would benefit z88dk as well.

zx70, 2021-03-06:
Thank you for getting in touch! 
It's possible we discussed it already,  but a couple of years ago I put some effort in disassembling the pc88 roms.
The results are here:
https://github.com/z88dk/techdocs/tree/master/targets/pc8801

bferguson3, 2021-03-07:
That's excellent, thank you! I will look into it asap.

(If there is an easier or more efficient discussion forum please let me
know. )

zx70, 2021-03-07:
I think this topic is fine.
If you wish something more structured we could add a topic in the z88dk.org forum

zx70, 2021-03-07:
https://z88dk.org/forum/viewforum.php?f=33

;)

bferguson3, 2021-03-07:
Thanks Stefano! Made a topic there.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

zx70, 2021-03-10:
#996
https://z88dk.org/forum/viewtopic.php?f=33&t=11462

closed 2021-03-10 by zx70