z88dk/z88dk #1712: Error in Winmake 2.1

fraespre, 2021-03-07:
Hi,

I use in zx target a custom memory map for sp1. 
In the realease 2.0 , it works without problems ( "Winmake zx" ):

```
WINMAKE: GENERATING Z80 LIBRARIES

target = zx
ZX7: Optimal LZ77/LZSS compression by Einar Saukas
File converted from 13400 to 5887 bytes! (delta 4)
ZX7: Optimal LZ77/LZSS compression by Einar Saukas
File converted from 13400 to 5886 bytes! (delta 3)
  zx_macro.m4
  zx_sccz80.lib
Se han movido         1 archivos.
  zx_sdcc_ix.lib
Se han movido         1 archivos.
  zx_sdcc_iy.lib
Se han movido         1 archivos.

C:\mio\soft\dev\z80\z88dk-2.0\libsrc\_DEVELOPMENT>
```

However, when I try the same in the new release 2.1 , it appears several errors ( "Winmake zx" ):

```
WINMAKE: GENERATING Z80 LIBRARIES

target = zx
ZX7: Optimal LZ77/LZSS compression by Einar Saukas
File converted from 13400 to 5887 bytes! (delta 4)
ZX7: Optimal LZ77/LZSS compression by Einar Saukas
File converted from 13400 to 5886 bytes! (delta 3)
  zx_macro.m4
  zx_sccz80.lib
Error at file 'arch/zx/bifrost_h/bifrost_h_asm.lst' line 8: cannot read file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE'
Error at file 'arch/zx/bifrost_l/bifrost_l_asm.lst' line 8: cannot read file 'arch/zx/bifrost_l/z80/BIFROSTL_ENGINE'
Error at file 'arch/zx/bifrost2/bifrost2_asm.lst' line 1: cannot read file 'arch/zx/bifrost2/z80/BIFROST2_INSTALL'
Error at file 'arch/zx/nirvanam/nirvanam_asm.lst' line 7: cannot read file 'arch/zx/nirvanam/z80/NIRVANAM_ENGINE'
Error at file 'arch/zx/nirvanap/nirvanap_asm.lst' line 7: cannot read file 'arch/zx/nirvanap/z80/NIRVANAP_ENGINE'
El sistema no puede encontrar el archivo especificado.
  zx_sdcc_ix.lib
Error at file 'arch/zx/bifrost_h/bifrost_h_asm.lst' line 8: cannot read file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE'
Error at file 'arch/zx/bifrost_l/bifrost_l_asm.lst' line 8: cannot read file 'arch/zx/bifrost_l/z80/BIFROSTL_ENGINE'
Error at file 'arch/zx/bifrost2/bifrost2_asm.lst' line 1: cannot read file 'arch/zx/bifrost2/z80/BIFROST2_INSTALL'
Error at file 'arch/zx/nirvanam/nirvanam_asm.lst' line 7: cannot read file 'arch/zx/nirvanam/z80/NIRVANAM_ENGINE'
Error at file 'arch/zx/nirvanap/nirvanap_asm.lst' line 7: cannot read file 'arch/zx/nirvanap/z80/NIRVANAP_ENGINE'
El sistema no puede encontrar el archivo especificado.
  zx_sdcc_iy.lib
Error at file 'arch/zx/bifrost_h/bifrost_h_asm.lst' line 8: cannot read file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE'
Error at file 'arch/zx/bifrost_l/bifrost_l_asm.lst' line 8: cannot read file 'arch/zx/bifrost_l/z80/BIFROSTL_ENGINE'
Error at file 'arch/zx/bifrost2/bifrost2_asm.lst' line 1: cannot read file 'arch/zx/bifrost2/z80/BIFROST2_INSTALL'
Error at file 'arch/zx/nirvanam/nirvanam_asm.lst' line 7: cannot read file 'arch/zx/nirvanam/z80/NIRVANAM_ENGINE'
Error at file 'arch/zx/nirvanap/nirvanap_asm.lst' line 7: cannot read file 'arch/zx/nirvanap/z80/NIRVANAP_ENGINE'
El sistema no puede encontrar el archivo especificado.

C:\mio\soft\dev\z80\z88dk-2.1\libsrc\_DEVELOPMENT>
```

Reference link with the instruction:
https://github.com/z88dk/z88dk/wiki/sp1_ex1

suborb, 2021-03-08:
I have to admit I've never used that file, and unfortunately you've ended up with a snapshot between #1481 and #1698

What this means is that in 2.1 you'll have to patch the Winmake.bat file yourself.

Take this change: a6cf639e65da3b033382ed0d6dbd7c06796015d9 and copy out the m4 invocations into the relevant bits of the batch file

fraespre, 2021-03-09:
Hi @suborb ,

In your comment you suggest me to apply the changes for the 'Makefile' file, but I'm working in Windows.
I have got the last version (master branch) of 'Winmake.bat' (with changes 28 days ago) and it works !!!

`C:\mio\soft\dev\z80\z88dk-2.1\libsrc\_DEVELOPMENT>Winmake zx

WINMAKE: GENERATING Z80 LIBRARIES

target = zx
"z88dk-zx0" no se reconoce como un comando interno o externo,
programa o archivo por lotes ejecutable.
"z88dk-zx0" no se reconoce como un comando interno o externo,
programa o archivo por lotes ejecutable.
  zx_macro.m4
  zx_sccz80.lib
Se han movido         1 archivos.
  zx_sdcc_ix.lib
Se han movido         1 archivos.
  zx_sdcc_iy.lib
Se han movido         1 archivos.

C:\mio\soft\dev\z80\z88dk-2.1\libsrc\_DEVELOPMENT>`